rauc/rauc #388: Slot already mounted detection fails with symlinks

xyzzy42, 2018-12-19:
I've configured a slot to have a device path which is a symlink, such as `/dev/disk/by-partlabel/rootfs0`.  Rauc tries to determine if this slot is already mounted, but it does this by comparing the device name, like above, with a list of mounted device from `g_unix_mounts_get()`.  But that list will use a device name like `/dev/mmcblk1p2`, since it's not created by the path which was mounted, but rather reconstructed from the major/minor of the device.  A simple string comparison of the two paths fails, even though they are the same device.

This results in `determine_slot_states()` failing to detect the existing mount and set `ext_mount_point`.  Later, `load_slot_status_locally()` will try to remount the already mounted filesystem and fail.

xyzzy42, 2018-12-19:
A solution to this would be to normalize device paths before comparing them.  This could be done by recursively following symlinks, i.e. realpath(), but that does not catch a number of cases.

The mounted device could be a file which is loopback mounted (which is already detected by rauc).  This file could have multiple hard links to it.  That would not be detected by comparing realpaths.

There is no rule that only one device node of a given type/major/minor combo can exist.  It's perfectly possible to mknod multiple nodes for the same underlying device.  This, much like multiple hardlinks to a file, it not detected by comparing realpaths.

Here's a solution which allows things that can be mounted to be compared in a way that is immune to the above and not terribly complex.

```
// Something that is, or can be, mounted onto a mount point
typedef struct {
        gboolean is_device; // vs being a loop mounted file
        dev_t dev;    // the device, or for a file, the device the file is on
        ino_t inode;  // inode of file for a non-device
} MountableObj;

// Take a device (or file) path and normalize it
static MountableObj *normalize_mountable_object(const gchar *devicepath)
{
        MountableObj *obj = g_new0(MountableObj, 1);
        GStatBuf st;

        if (g_stat(devicepath, &st) == -1) {
                /* Virtual filesystems like devpts trigger this case */
                g_debug("Can't stat '%s', must not be mountable", devicepath);
                return NULL;
        }

       if (S_ISBLK(st.st_mode)) {
               obj->is_device = TRUE;
               obj->dev = st.st_rdev;
       } else if (S_ISREG(st.st_mode)) {
               obj->is_device = FALSE;
               obj->dev = st.st_dev;
               obj->inode = st.st_ino;
       } else {
               g_debug("Device '%s' is not something which is mountable", devicepath);
               g_clear_pointer(&obj, g_free);
       }
       return obj;
}

/* Compare two MountableObj for equality */
static gboolean is_same_mountable_object(const MountableObj *a, const MountableObj *b)
{
       return a->is_device == b->is_device && a->dev == b->dev &&
               (a->is_device || a->inode == b->inode);
}

```

xyzzy42, 2019-01-02:
I could send a pull request to fix this in the manner I described above if it appears acceptable.

jluebbe, 2019-01-28:
@xyzzy42 Your approach seems better than what @ejoerns tried in #395. I'd suggest to pick up the virtual fs check (if it doesn't start with a slash, we don't want to try following random symlinks in the CWD):
```
		if (m_devpath == NULL || m_devpath[0] != '/')
￼				continue;
```

xyzzy42, 2019-02-11:
There's an issue getting the tests to work.

`find_slot_by_device`/`find_config_slot_by_device` are used twice in rauc's code, both times on a running target system where we want to know if a real device corresponds to a slot.

It's also used over a dozen times in the test code.  It doesn't work because the devices, both the one used to look up a slot and the device named in the slot, do not exist.  On a target, we couldn't say if they are the same or not.  So the lookup fails.

It looks to me like `find_config_slot_by_device()` was being used simple as away to get a handle to slot.  It wasn't intended to test the by device lookup.  Maybe a better way would be to create a `find_slot_by_name()` function and use that in the test code?