rauc/rauc #389: Using RAUC to update FS

hikboy, 2018-12-20:
I'm using rauc to update my fs. 
And I have a problem about the installing of it.
I need to do sth to the fs, when rauc does mount my partition and copys tarball to the mountpoin is ok, and before does umount it.
I want to know if rauc has the mechanism or function.
Or, can I run a script between "09.52.47" and "09.54.11".

09.52.47 rauc-Message: Mounting ext4 slot /dev/mmcblk0p3
**09.52.47** rauc-Message: Extracting /mnt/rauc/bundle/rootfs-arm64.tar.bz2 to /mnt/rauc/rootfs.0
**09.54.11** rauc-Message: Unmounting ext4 slot /dev/mmcblk0p3
09.54.37 rauc-Message: installing /mnt/rauc/upt/uptst.raucb: Updating slot rootfs.0 status
09.54.37 rauc-Message: Updating slot file /mnt/rauc/rootfs.0/slot.raucs
09.54.37 rauc-Message: Marking slots as bootable...
09.54.37 rauc-Message: installing /mnt/rauc/upt/uptst.raucb: Updating slot rootfs.0 done
09.54.38 rauc-Message: Updating slot file /mnt/rauc/rootfs.0/slot.raucs
09.54.38 Starting post install handler: /etc/rauc/post_script.sh
09.54.38 rauc-Message: installing /mnt/rauc/upt/uptst.raucb: All slots updated

Bastian-Krause, 2018-12-20:
Hi!

Have a look at the [slot hooks documentation](https://rauc.readthedocs.io/en/latest/using.html#slot-hooks):

> The post-install hook will be called right after the update procedure for the respective slot was finished successfully. For slot types that represent a mountable file system, the hook will be executed with having the file system mounted. This allows to write some post-install information to the slot.

Bastian

hikboy, 2018-12-20:
Oh! 
Thanks for your help. 
It's my own fault that ignored the information.

closed 2018-12-20 by hikboy