rauc/rauc #394: Verify error:unable to get local issuer certificate

sholzmayer, 2019-01-18:
The installation of the self signed raucb file always fails with the following error message: `Verify error:unable to get local issuer certificate`.

The output of `rauc info my-file.raucb` is:
```
rauc-Message: 13:57:57.329: Reading bundle: my-file.raucb
rauc-Message: 13:57:57.401: Verifying bundle... 
signature verification failed: Verify error:unable to get local issuer certificate
```

Any idea what the problem could be? Where is the `local issuer certificate` located?

sholzmayer, 2019-01-19:
Yeah in the `/etc/rauc/system.conf` is the keyring defined and also present on the system:
```
[keyring]
path=/etc/rauc/keyring.pem
```
Any idea about the `local issuer certificate`? What is it and where is it located at?

ejoerns, 2019-01-24:
@sholzmayer did you resolve your issue?

sholzmayer, 2019-01-25:
This solved my problem. Thank you very much!

I used the wrong certificate combination, because the CA also had a root certificate and i got a certificate mismatch.

closed 2019-01-25 by sholzmayer