rauc/rauc #396: Running rauc services fails

Sharvin26, 2019-01-23:


ejoerns, 2019-01-23:
What is the type of Linux you test this on? Do you use systemd or initv?

Could you provide the log of at least `make install` run?

Does RAUC reveal anything else than this message? Does `rauc service -d` produce more details?

Sharvin26, 2019-01-23:
I use systemd
Output for rauc service -d 
```
rauc-Message: Debug log domains: 'rauc'

(rauc:12703): rauc-DEBUG: service start
(rauc:12703): rauc-DEBUG: system config not found, using default values
Failed to obtain name de.pengutronix.rauc
```

Output for make install
```
make --no-print-directory install-am
 /bin/mkdir -p '/usr/local/bin'
  /bin/bash ./libtool   --mode=install /usr/bin/install -c rauc '/usr/local/bin'
libtool: install: /usr/bin/install -c rauc /usr/local/bin/rauc
 /bin/mkdir -p '/usr/local/share/dbus-1/system.d'
 /usr/bin/install -c -m 644 data/de.pengutronix.rauc.conf '/usr/local/share/dbus-1/system.d'
 /bin/mkdir -p '/usr/local/share/dbus-1/system-services'
 /usr/bin/install -c -m 644 data/de.pengutronix.rauc.service '/usr/local/share/dbus-1/system-services'
 /bin/mkdir -p '/usr/local/libexec'
 /usr/bin/install -c data/rauc-service.sh '/usr/local/libexec'
 /bin/mkdir -p '/lib/systemd/system'
 /usr/bin/install -c -m 644 data/rauc.service '/lib/systemd/system'
```

Sharvin26, 2019-01-24:
This is a Brief procedure I have followed and My Understanding of whole Ota Scenario :
1. First of all, right now I am targeting only file update no operating system update.
2.  I am using Bosch IoT Rollout suite as my Host which is compatible with the Rauc Hawkbit and Rauc as Installer [Bosch IoT Rollout Documentation](https://docs.bosch-iot-rollouts.com/documentation/introduction/ecosystem.html).
3. I am using Raspberry Pi as my target. I have installed the Debian operating system and is based on systemd.
4. I started following this [documentation](https://github.com/rauc/rauc-hawkbit). I faced some problem while configuring this file on my system. Steps are as follows:
- Installed all the Dependencies for the System related to the packages. 
- Faced an issue related to the `aiohttp.Timeout` in `/lib/python3.5/site-packages/rauc-hawkbit/ddi/client.py`.
`AttributeError: module 'aiohttp' has no attribute 'Timeout'`.
- Followed this [url](https://github.com/GNS3/gns3-server/issues/463) for fixing the issue by replacing `aiohttp.Timeout` with `async_timeout.timeout`.
- And my rauc hawkbit execution file started running properly till the Installation procedure:
```
    DEBUG    POST https://url//TenantId/controller/v1/controllerId/deploymentBase/..../feedback
    WARNING  Polling failed with a temporary error: g-dbus-error-quark: GDBus.Error:org.
    freedesktop.DBus.Error.ServiceUnknown: The name de.pengutronix.rauc was not provided by 
    any .service files (2)
    INFO     Retry will happen in 60 seconds
```
5. As I started facing this issue we had a discussion related to the rauc on this [issue](https://github.com/rauc/rauc-hawkbit/issues/15). So I followed the rauc [Installation procedure](https://github.com/rauc/rauc)
- For Required kernel options in my `/boot/config`:
I configured 
```
CONFIG_BLK_DEV_LOOP=y
CONFIG_SQUASHFS=y
```
- And After that:
```
sudo apt-get install automake libtool libglib2.0-dev libcurl3-dev libssl-dev
sudo apt-get install libjson-glib-dev
git clone https://github.com/rauc/rauc
cd rauc
./autogen.sh
./configure
make
make install
``` 
- Output for rauc make install:
```
make --no-print-directory install-am
 /bin/mkdir -p '/usr/local/bin'
  /bin/bash ./libtool   --mode=install /usr/bin/install -c rauc '/usr/local/bin'
libtool: install: /usr/bin/install -c rauc /usr/local/bin/rauc
 /bin/mkdir -p '/usr/local/share/dbus-1/system.d'
 /usr/bin/install -c -m 644 data/de.pengutronix.rauc.conf '/usr/local/share/dbus-1/system.d'
 /bin/mkdir -p '/usr/local/share/dbus-1/system-services'
 /usr/bin/install -c -m 644 data/de.pengutronix.rauc.service '/usr/local/share/dbus-1/system-services'
 /bin/mkdir -p '/usr/local/libexec'
 /usr/bin/install -c data/rauc-service.sh '/usr/local/libexec'
 /bin/mkdir -p '/lib/systemd/system'
 /usr/bin/install -c -m 644 data/rauc.service '/lib/systemd/system'
```
6. Once the rauc was properly installed using make install I tried `rauc service` and it gave me following output:
`Failed to obtain name de.pengutronix.rauc`
7. I started refering this [Rauc Integration documentation](https://rauc.readthedocs.io/en/latest/integration.html#id9)
According to the documentation I had few doubts
- I couldn't find rauc directory under `/etc/` nor I could find `/system.conf` which is placed in `/etc/rauc/system.conf`
- Regarding the slot configuration as I am only considering the **file update** do I need to specify the slot configuration ?
- For Kernel configuration I couldn't find Kconfig but I found a path `/boot/config` Where I found these 2 options and I configured them as follows.
```
CONFIG_BLK_DEV_LOOP=y
CONFIG_SQUASHFS=y
```
- Is it necessay to Interface with bootloader as I am only working with file update.
- For Systemd integration I found `rauc.service` in `/data` which by default my make install as placed in `\lib\systemd\system\`
- For Dbus Integration as per the documentation **properly install the D-Bus service file de.pengutronix.rauc.service** So I started searching and I found that dbus services are located in `/usr/share/dbus-1/services/`. This directory was missing `de.pengutronix.rauc.service` So i copied the file from `/rauc/data/` directory to that path. Tried reloading the dbus service and deamon still no change.
- Tried ptxdist but I could not find a ptxidist menu_config file. Thus facing that error.
6. For Rauc Status 
```
(rauc:12874): GLib-CRITICAL **: g_hash_table_destroy: assertion 'hash_table != NULL' failed
rauc-Message: rauc mark: rauc status: error retrieving slot status via D-Bus: error creating proxy: Error calling StartServiceByName for de.pengutronix.rauc: Timeout was reached
```

ejoerns, 2019-01-24:
@Sharvin26 As you might have seen, `make install` uses prefix `/usr/local`. If you want to have it installed in `/usr`, you should try `./configure --prefix=/usr` and then run `make && make install` again.

Anyway, embedded Linux build systems will do that all for you and you will need them, anyway for a proper integration of things.

For starting with PTXdist, reading documentation or using DistroKit as a base is a good idea: https://www.pengutronix.de/de/blog/2017-08-28-DistroKit_Intro.html

ejoerns, 2019-01-24:
The `g_hash_table_destroy` assertion error however appears to be a bug in RAUC's cleanup handling.

Sharvin26, 2019-01-24:
I tried 
```
./configure --prefix=/usr
make 
sudo make install
``` 
log of make install
```
 /bin/mkdir -p '/usr/bin'
  /bin/bash ./libtool   --mode=install /usr/bin/install -c rauc '/usr/bin'
libtool: install: /usr/bin/install -c rauc /usr/bin/rauc
 /bin/mkdir -p '/usr/share/dbus-1/system.d'
 /usr/bin/install -c -m 644 data/de.pengutronix.rauc.conf '/usr/share/dbus-1/system.d'
 /bin/mkdir -p '/usr/share/dbus-1/system-services'
 /usr/bin/install -c -m 644 data/de.pengutronix.rauc.service '/usr/share/dbus-1/system-services'
 /bin/mkdir -p '/usr/libexec'
 /usr/bin/install -c data/rauc-service.sh '/usr/libexec'
 /bin/mkdir -p '/lib/systemd/system'
 /usr/bin/install -c -m 644 data/rauc.service '/lib/systemd/system'
```
Then 
```
~ rauc service
Failed to obtain name de.pengutronix.rauc
```
Also tried the ptxdist https://www.pengutronix.de/de/blog/2017-08-28-DistroKit_Intro.html
For this command I am getting 
```
~ sudo apt install oselas.toolchain-2016.06.1-arm-v7a-linux-gnueabihf-gcc-5.4.0-glibc-2.23-binutils-2.26-kernel-4.6-sanitized
E: Type 'http://debian.pengutronix.de/debian/' is not known on line 1 in source list /etc/apt/sources.list.d/pengutronix.sid.list
E: The list of sources could not be read.
E: Type 'http://debian.pengutronix.de/debian/' is not known on line 1 in source list /etc/apt/sources.list.d/pengutronix.sid.list
E: The list of sources could not be read.
```
I also tried meta rauc I have successfully configured bitbake and openembeddedcore in my system.
I am following this [documentation](https://github.com/rauc/meta-rauc) But I am confused on this line

> 
4. Build a bundle and the rootfs for your device
  bitbake my-bundle-recipe

What would be the bundle-recipe which I need to build I tried building meta-rauc directory it failed then I tried building `~/meta-rauc/recipes-core/bundles/core-bundle-minimal.bb` and it's giving me the error
```
ERROR: Nothing PROVIDES 'core-bundle-minimal.bb'. Close matches:
  core-image-minimal
```

ejoerns, 2019-01-24:
@Sharvin26 Ahm, did you probably try to start RAUC as a user? Of course this needs to be run as root!

That's the last guess I have. Despite it is very unlikely that one runs a core service as user on an embedded system, bust as I am not sure what your setup is so I would assume this to be the issue possibly.

About the PTXdist try: I fear the blog post lacks a `deb` in the echo command. Thus it should be:

    sudo echo 'deb http://debian.pengutronix.de/debian/ sid main contrib non-free' > /etc/apt/sources.list.d/pengutronix.sid.list

About the `meta-rauc` try: Are you sure you added meta-rauc to you `conf/bblayers.conf`?

Sharvin26, 2019-01-25:
1. I tried running Rauc as root. 
For `rauc service` I almost waited for 4 to 5 minutes Still I didn't receive any output.
2. For PTXdist it's giving me error's related to Unsigned Public key.
3. For meta-rauc Thanks, I configured the BBpath properly
Now I am getting the error `layer is compatible with thud rocko sumo`
Full Log of error 
```
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/home/debian/meta-rauc/recipes-core/bundles/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 4415 at 2019-01-25 06:40:45.598874 ---
Traceback (most recent call last):
  File "/home/debian/bitbake/lib/bb/cookerdata.py", line 284, in parseBaseConfiguration
    self.data = self.parseConfigurationFiles(self.prefiles, self.postfiles)
  File "/home/debian/bitbake/lib/bb/cookerdata.py", line 398, in parseConfigurationFiles
    % (c, " ".join(layerseries), " ".join(compat)))
  File "/home/debian/bitbake/lib/bb/__init__.py", line 120, in fatal
    raise BBHandledException()
bb.BBHandledException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/debian/bitbake/lib/bb/daemonize.py", line 83, in createDaemon
    function()
  File "/home/debian/bitbake/lib/bb/server/process.py", line 469, in _startServer
    self.cooker = bb.cooker.BBCooker(self.configuration, self.featureset)
  File "/home/debian/bitbake/lib/bb/cooker.py", line 210, in __init__
    self.initConfigurationData()
  File "/home/debian/bitbake/lib/bb/cooker.py", line 375, in initConfigurationData
    self.databuilder.parseBaseConfiguration()
  File "/home/debian/bitbake/lib/bb/cookerdata.py", line 317, in parseBaseConfiguration
    raise bb.BBHandledException
bb.BBHandledException
ERROR: Layer rauc is not compatible with the core layer which only supports these series:  (layer is compatible with rocko thud sumo)
```

ejoerns, 2019-01-25:
1. I think that is the intended behavior. Or what kind of output do you expect. It will print some, if you start it with `-d`. It's a daemon process, it is designed to start and keep running.

Try interacting with it (over D-Bus) by calling `rauc status` for example.

2. Too less information

3. Seems to be a case for reading the first lines of a Yocto guide. How did you start? When cloning poky, you should have oe-core in you bblayers.conf by default...

Sharvin26, 2019-01-25:
1. Output for rauc status and rauc service -d
```
root:/home/debian# rauc status
(rauc:2605): rauc-WARNING **: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code30: Failed getting primary slot: Obtaining primary entry from bootloader '(null)' not supported yet
Compatible:
Variant:
Booted from: /dev/mmcblk1p1
Activated:   (null) ((null))
slot states:

root:/home/debian# rauc service -d
rauc-Message: Debug log domains: 'rauc'
(rauc:2612): rauc-DEBUG: service start
(rauc:2612): rauc-DEBUG: system config not found, using default values
Failed to obtain name de.pengutronix.rauc
```
3. What i have did is cloned bitbake and OE.
Followed this guide to configure both 
- [OE config](https://www.openembedded.org/wiki/OE-Core_Standalone_Setup)
- [BitBake](https://wiki.gumstix.com/index.php/Bitbake_on_Ubuntu)
Configured Both and then I followed the [meta rauc guide](https://github.com/rauc/meta-rauc) 
Some Steps were missing in th guide so I found them through various sources
```
cd /meta-rauc/recipes-core/bundles
```
It was giving me error related to the conf directory so I copied the directory here.
In conf directory there was only a layer.conf and it was giving me missing file error related to the bblayers.conf and bitbake.conf.So I created those file.

BBlayers.conf
```
BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/debian/oe-core/meta \
  "
```

bitbake.conf
```
TMPDIR  = "${TOPDIR}/tmp"
CACHE   = "${TMPDIR}/cache"
STAMP   = "${TMPDIR}/stamps"
T       = "${TMPDIR}/work"
B       = "${TMPDIR}"
```

layer.conf
```
# We have a conf and classes directory, add to BBPATH
BBPATH .= ":${LAYERDIR}"
# We have recipes-* directories, add to BBFILES
BBFILES += "${LAYERDIR}/recipes-*/*/*.bb \
	${LAYERDIR}/recipes-*/*/*.bbappend"
BBFILE_COLLECTIONS += "rauc"
BBFILE_PATTERN_rauc = "^${LAYERDIR}/"
BBFILE_PRIORITY_rauc = "6"
LAYERDEPENDS_rauc = "core"
# meta-python is needed to build/run hawkbit-client
LAYERRECOMMENDS_rauc = "meta-python"
# meta-filesystems is needed to build cascync with fuse support (the default)
LAYERRECOMMENDS_rauc += "meta-filesystems"
LAYERSERIES_COMPAT_rauc = "sumo rocko thud"
```
And now I am trying to run
` cd meta-rauc/recipes-core/bundles`
` bitbake core-bundle-minimal.bb`
And the error I am getting is
```
Loading cache: 100% |##########################################################################################################################| Time: 0:00:07
Loaded 1255 entries from dependency cache.
ERROR: Nothing PROVIDES 'core-bundle-minimal.bb'. Close matches:
  core-image-minimal

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```
Can you please give me some defined path for Configuring Rauc Installer on Debian Os ?

Sharvin26, 2019-01-25:
Am I do anything wrong in the configuration? It would be great if you can give a detailed description of the building and configuring the bitbake, Oe core and Rauc installer.

Sharvin26, 2019-01-25:
For Oe core environment from this documentation for this command
`bitbake core-image-minimal`
I get 
```
Sstate summary: Wanted 466 Found 0 Missed 466 Current 0 (0% match, 0% complete)
NOTE: Executing SetScene Tasks
WARNING: The free space of /home/debian/oe-core/build/downloads (/dev/mmcblk1p1) is running low (0.697GB left)
ERROR: No new tasks can be executed since the disk space monitor action is "STOPTASKS"!
WARNING: The free space of /home/debian/oe-core/build/tmp-glibc (/dev/mmcblk1p1) is running low (0.697GB left)
ERROR: No new tasks can be executed since the disk space monitor action is "STOPTASKS"!
WARNING: The free space of /home/debian/oe-core/build/sstate-cache (/dev/mmcblk1p1) is running low (0.697GB left)
ERROR: No new tasks can be executed since the disk space monitor action is "STOPTASKS"!
NOTE: Tasks Summary: Attempted 3 tasks of which 3 didn't need to be rerun and all succeeded.
```

As I have a memory limitation How can this help in the client side?

Sharvin26, 2019-01-25:
I had one more doubt as bitbake is building a build When I do it for meta rauc will it create a new system image or it will configure the Rauc Installer on my existing Debian Operating system. Just had the Querry in the mind.

Sharvin26, 2019-01-25:
For Configuring the slots for rauc Hawkbit What should I do it fails at that point
```
2019-01-25 11:55:08 INFO     Update progress: 0% Determining slot states
Progress:   0% - Determining slot states
2019-01-25 11:55:08 DEBUG    POST https://device.eu1.bosch-iot-rollouts.com//<Tennant Id>/controller/v1/<device-Id>/deploymentBase/..../feedback
2019-01-25 11:55:09 INFO     Update progress: 20% Determining slot states failed.
Progress:  20% - Determining slot states failed.
```

ejoerns, 2019-01-25:
https://rauc.readthedocs.io/en/latest/examples.html#example-slot-configurations

Sharvin26, 2019-01-25:
Where should this configuration be added?

Sharvin26, 2019-01-25:
For the update file or target device?
I have added the test cases log for the devices it's failing can you once check them and tell me what would be the issue.
Even I am unable to still find /etc/rauc directory in the target device.
Also, rauc service is failing

ejoerns, 2019-01-25:
@Sharvin26 /etc/rauc/ will not appear by default (unless you use PTXdist or Yocto/OE), but you can simply create a dir with this name and place your system.conf there.

system.conf is required for the target only. On the host side, neither D-Bus stuff nor service nor configuration is required. The host rauc tool only handles bundle creation and inspection

Sharvin26, 2019-01-28:
Okay, So I need to configure Rauc at the Host Side also? And for configuration purpose, I'll need to use the same process right? I had a doubt does rauc only support Linux or it's cross-platform.

ejoerns, 2019-01-28:
Not sure if I get your question right.
You need RAUC on the host side, too. At least to create the update bundles.

And yes, RAUC only supports Linux. Only. Intentionally.

Was that your question? When developing for Linux, using a Linux host system (or at least VM) is mandatory imho..

Sharvin26, 2019-01-28:
Yes, that was the Question. I need to create an update bundle so that will be created using Linux machine. For Host side, I am using Bosch IoT Suite So I'll be needing Linux system only for creating the Update bundle to upload it on the IoT rollout suite. Is my understanding correct?

Sharvin26, 2019-01-28:
What does Compatible miss match mean?
```
2019-01-28 07:44:26 INFO     Update progress: 100% Installing failed.
Progress: 100% - Installing failed.
2019-01-28 07:44:26 DEBUG    POST https://device.eu1.bosch-iot-rollouts.com//<Tennant-id>/controller/v1/<device-id>/deploymentBase/.../feedback
2019-01-28 07:44:27 INFO     Last error: Installation error: Compatible mismatch
2019-01-28 07:44:27 DEBUG    POST https://device.eu1.bosch-iot-rollouts.com//<Tennant-id>/controller/v1/<device-id>/deploymentBase/.../feedback
2019-01-28 07:44:27 INFO     Rauc bundle update completed with result: 1
2019-01-28 07:44:27 DEBUG    https://device.eu1.bosch-iot-rollouts.com//<Tennant-id>/controller/v1/<device-id>/deploymentBase/.../feedback
Result:   FAILED
```
I have kept same name for compatible in my `/etc/rauc/system.conf` and `/temp-dir/manifest.raucm`.
```
[system]
compatible=rauc-demo
bootloader=grub
mountprefix=/mnt/rauc

[keyring]
path=demo.cert.pem

[slot.rootfs.0]
device=/dev/sda1
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/sda2
type=ext4
bootname=B
```
While my manifest-raucm 
```
[update]
compatible=rauc-demo
version=2015.04-1

[image.rootfs]
sha256=...
size=23
filename=update.text
```

Sharvin26, 2019-01-28:
I am getting the error related to the Grub configuration:
```
2019-01-28 08:09:44 INFO     Update progress: 100% Installing failed.
Progress: 100% - Installing failed.
2019-01-28 08:09:44 DEBUG    POST https://device.eu1.bosch-iot-rollouts.com//<Tennant-ID>/controller/v1/<Device-Id>/deploymentBase/..../feedback
2019-01-28 08:09:45 INFO     Last error: Installation error: Failed marking slot rootfs.1 non-bootable: Failed marking 'rootfs.1' as bad: Failed to start grub-editenv: Failed to execute child process "grub-editenv" (No such file or directory)
2019-01-28 08:09:45 DEBUG   POST https://device.eu1.bosch-iot-rollouts.com//<Tennant-ID>/controller/v1/<Device-Id>/deploymentBase/..../feedback
2019-01-28 08:09:45 INFO     Rauc bundle update completed with result: 1
2019-01-28 08:09:45 DEBUG    POST https://device.eu1.bosch-iot-rollouts.com//<Tennant-ID>/controller/v1/<Device-Id>/deploymentBase/..../feedback
Result:   FAILED
```
I read the [Documentation](https://rauc.readthedocs.io/en/latest/examples.html). I couldn't find `/boot/grub/grubenv` in my debian Operating System and I only have one rootfs partition How can I add rootfs to another partition for /dev/sda2.

ejoerns, 2019-01-28:
@Sharvin26 Sorry, but I cannot offer basic system integration support here. And developing a redundant update solution is far more than just compiling RAUC (as you already noticed).

This is still a RAUC open source software community channel. I guess what you are looking for is professional business support, such as my company Pengturonix or other companies offer. This here exceeds what I can and am willing to answer on in my spare time. I hope you can accept this.

I am willing to still answer questions targeting issues with RAUC, also some slightly build system-specific things. But I will not provide a step-by-step interactive setup guide that could be replaced by reading the very basics of how to build an embedded system. Hope you understand this.

On you last questions: I have never used Bosch IoT Suite suite so far. I do not know anything about it.

Your last question triggers  the consideration on my side if an imaged-based update solution is what you are looking for?... https://rauc.readthedocs.io/en/latest/basic.html#rauc-basics

ejoerns, 2019-01-30:
@Sharvin26 RAUC is designed to work with archives or images. Lot's of the way it handles updates deals with inspecting file extensions, slot types, etc.
The more important, RAUC is designed for redundant atomic updates. I am not sure how you make this fit your use case. You could tell rauc to install applications into different dirs and use a symlink or whatever as an atomic switch between these..

Of course you can use RAUC as a container for deploying updates and implement all the install logic in a custom handler.. But not sure if you gain any benefit from this.

What about using standard package management, or something like [os-tree](https://ostree.readthedocs.io/en/latest/manual/introduction/)?

ejoerns, 2019-01-30:
Note: As a result of this and other discussions I've created #399

Sharvin26, 2019-01-31:
Okay, Thank you, sir, I'll check it.

Sharvin26, 2019-02-13:
Hello Sir, As per the discussion from the above thread I created a custom Hook for my Bundle.

I am working on A+B redundancy update Where I am keeping one copy of my operating system on the emcc while other will be downloaded and installed on the sd card.

Steps and Configuration I am following are:
1. My `system.conf` under /etc/rauc/ is
```
[system]
compatible=rauc-demo
bootloader=uboot

[keyring]
path=cert.pem

[slot.rootfs.0]
device=/dev/sda1
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/sdb1
type=ext4
bootname=B

[slot.cfg.1]
device=/dev/sdb1
type=ext4
parent=rootfs.1
```
2. My `manifest.raucm`:
```
[update]
compatible=rauc-demo

[hooks]
filename=hook.sh

[image.rootfs]
filename=rootfs.img

[image.cfg]
filename=update.tar
hooks=post-install;
```
3. My `hook.sh`:
```
#!/bin/sh
echo "I am at the start of shell script."
case "$1" in
  slot-post-install)
    echo "I am above Rauc_Slot_class case."
    case "$RAUC_SLOT_CLASS" in
      cfg)
        echo "Unziping the Operating Systems."
        unxz /update.img.xz
        echo "Unzippedd the Operating System. Installing th update!!!"
        dd if=/update.img bs=4M of=/dev/sdb
        echo "Installation done"
        ;;
      *)
        echo "Internal error: hook mismatched"
        exit 11
    esac
    ;;
  *)
    echo "Internal error: unrecognized hook"
    exit 11
    ;;
esac

exit 0
```

I am sending a `update.tar` which rauc internally decompresses into `update.img.xz` which I am converting to `update.img` using a unxz tool and then installing the update using dd.
Now When I fire a command `rauc -d install update.raucb` It successfully verifies the bundle, runs the hook shell script, installs the update. I debugged using `journalctl -u rauc` and I came across that it executes till the debug line `Installation done` from the above hook.sh. After that it print's the following log:

```
Unmounting ext4 slot /dev/sdb1
installing /home/os/update.raucb: Updating slot cfg.1 status
mount: wrong fs type, bad option, bad superblock on /dev/sdb1,
missing codepage or helper program, or other error
In some cases useful info is found in syslog - try
dmesg | tail or so.
Installation error: Error while writing status file: failed to mount slot: failed to run mount: Child process exited
installing /home/os/update.raucb: Installation error: Error while writing status file: failed to mount slot:
installing /home/os/update.raucb: finished
installing `/home/os/update.raucb` failed: 1
```
The `dmesg | tail ` log:
```
[ 1602.175105] loop: module loaded
[ 1602.261539] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[ 1602.609039] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[ 1602.785061] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[ 1650.257587] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[ 1656.494075] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[ 2492.122481] EXT4-fs error (device mmcblk0p1): ext4_iget:4723: inode #8: comm mount: checksum invalid
[ 2492.134919] EXT4-fs (sdb1): no journal found
[ 3426.835671] EXT4-fs error (device sdb1): ext4_iget:4723: inode #8: comm mount: checksum invalid
[ 3426.858823] EXT4-fs (sdb1): no journal found
```

So I tried installing the update manually using only `xz` and `dd` tool. I copied the `update.img.xz` on my `/dev/sdb1`.
After that, I did `unxz /update.img.xz` then `dd if='/update.img' bs=4M of=/dev/sdb`. I mounted the `/dev/sdb1` using `mount -t auto /dev/sdb1 osTest/`. And I found that my operating system was succesfully installed.

I am confused about why the error occurs when using the Rauc installer? As manually this method is working fine I am able to successfully flash the image on my sd card but fails when using the rauc installer.

ejoerns, 2019-02-13:
Not sure if I can follow you. But first of all, your configuration appears to be broken for me...

Why do you have A on eMMC and B on SD card?

Why is A different from B?

Maybe you confused about A + B setup and wanted a A+recovery?

Why are `rootfs.1` and `cfg.1` referencing the same device? 

What does your tar actually contain? The base system?

What do you mean by
> I am sending a update.tar which rauc internally decompresses into update.img.xz

.tar to .img.xz does not sound like 'decompression' and RAUC does not change image types, anyway. Thus where does this come from??

Sharvin26, 2019-02-13:
1.  Not sure if I can follow you. But first of all, your configuration appears to be broken for me...
=> I am trying to develop A + B redundancy setup. While reading the documentation and various implementations of other people I found this to be a suitable case for my implementation. Can you please help me where I might have done the configurations wrong? 

2. Why do you have A on eMMC and B on SD card?
=> As I have limited memory on my hardware and to implement A + B Setup I had to use the eMMC as the A Partition while SD card has a B partition. 

3. Why is A different from B?
=> A is an eMMC partition while B is an SD card partition.

4. Maybe you confused about A + B setup and wanted an A+recovery?
=> According to my understanding the update is performed on the inactive partition while the user can perform operations on the active partition. Once the installation is complete user needs to reboot and inactive partition is booted as the active partition.

5. Why are `rootfs.1` and `cfg.1` referencing the same device?
=> I found this in one of the issues from Rauc Github. As I didn't find `cfg.1` in any of the documentation ( or I might have missed it I guess ) According to my understanding as referenced from the documentation 

> A slot is defined by a slot section. The naming of the section must follow a simple format: [slot.<slot-class>.<slot-index>] where <slot-class> describes a class of possibly multiple redundant slots (such as rootfs, recovery or appfs) and slot-index is the index of the individual slot instance, starting with index 0.

So I am considering cfg needs parent class for the slot to work.

6. What does your tar actually contain? The base system?
=> My Tar contains an Operating system image ( As the image size is too large I had to compress it. )

7. What do you mean by .tar to .img.xz does not sound like 'decompression' and RAUC does not change image types, anyway. Thus where does this come from??
> I am sending a update.tar which rauc internally decompresses into update.img.xz

=> According to my understanding, Rauc uses squashFs. I referred this line from the documentation 
> To pack this all together, the default bundle format uses SquashFS. This provides good compression while allowing to mount the bundle without having to unpack it on the target system.

Please correct if anything's wrong from the above. 

I am unable to debug why the update is failing when using the rauc installer as I have applied the same logic on the hooks that I have already tested and verified when installing the operating system manually. I guess it's failing while mounting the partition. But I am unable to find the proper solution for the problem mentioned in the above thread. 
Thanks.

ejoerns, 2019-02-13:
Ok, you want to use a fully redundant A+B setup where A+B are different devices. This is the point I got now. And not because you do not trust you eMMC, but because you need more space than you have.

Thus I would recommend buying a different hardware.

Anyway, if we accept doing it this way. I still have no idea what `cfg.1` is.

> => I found this in one of the issues from Rauc Github. As I didn't find cfg.1 in any of the documentation ( or I might have missed it I guess ) According to my understanding as referenced from the documentation

Just finding something somewhere really does not have anything to do with targeting your actual problem...
Redundancy boot is not a work for copy & pasters.

The only issue I found where this is referenced is https://github.com/rauc/rauc/issues/224. But this has absolutely nothing to do with your current setup. They use a **redundant** and **separate** configuration data partition.
What you only want is

``` cfg
[slot.rootfs.0]
device=/dev/sda1
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/sdb1
type=ext4
bootname=B
```

Then your manifest: If you intend to install a tar into a rootfs slot, then you only have a **single** image for the slot class **rootfs**.

``` cfg
[image.rootfs]
filename=update.tar
hooks=post-install;
```

And there is **really really** nothing that converts your tar into something else. Yes, RAUC uses squashfs. Squashfs is the outer container of the bundle and a compressed file system. Once it is mounted all files in it appear as they would on each other file system, such as on your ext4 or whatever.

Thus if you have a update.tar in you bundle and your hook should handle it, simply refer to `update.tar` file name.

If the tar simply contains your system's image, then throw away your hook! RAUC does this by itself. It creates an ext4 for your and unpacks the tar into it...

If instead the tar contains your image and is then compresses or the tar compressed or .. mh.. did not get it..

1. tar is a an *archive* format, basically *without* any compression. .tar.{gz,xz,..} ar *compressed* tars.
2. Packing a compressed file system image in a tar makes no sense
3. Compressing your file system before packing it into the compressed bundle also makes no sense, because the maximum effect of double compression you will get is an increased size...
4. Compressing the update image also has nothing to do with limited resources on your target, unless you copy it to your target before you installing it. Which is *only* meaningfull if you download it via network. but then you can still put it into a RAM file system or use casync.

Thus what you probably actually want is

``` cfg
[image.rootfs]
filename=update.ext4
```

sigh

Sharvin26, 2019-02-14:
Yes, I am copying the update to the target system before installing as I am downloading the update over the network.
> 4\. Compressing the update image also has nothing to do with limited resources on your target, unless you copy it to your target before you installing it. Which is _only_ meaningfull if you download it via network. but then you can still put it into a RAM file system or use casync.

Thank you for your help.

ejoerns, 2019-03-21:
Any questions left here? Otherwise I will close this.

ejoerns, 2019-06-04:
I assume this to be resolved.

closed 2019-06-04 by ejoerns