rauc/rauc #398: Partition is always marked good

philippbachmann08, 2019-01-29:
I want to use the `rauc status mark-good` command manually at a specific point in the boot process, to decide whether I want to stay on that partition or switch it. Therefore I disabled the `rauc-mark-good` systemd service (even removed it from the device). However, when I reboot the system the `boot status` (by running `rauc status`) shows `good`. According to https://rauc.readthedocs.io/en/latest/using.html#react-to-a-successfully-booted-system-failed-boot I assume that it should show `bad`. 

How can I set it to `bad` until I run `rauc status mark-good` manually?

ejoerns, 2019-01-29:
@philippbachmann08 This depends on how you handle it in the bootloader.

By default, most implementations assume you have an attempts counter that is decreased before you boot and increased when booted. This is basically what `rauc status mark-good` does, too. Thus, even if you still get a `good` result after one boot, the status will turn bad after subsequent boots when no good-marking service ran.

If you want to enforce the number of attempts to be exactly `1`, then you simply disable the boot slot in the bootloader while booting. Then `rauc status` shoudl also show `bad`. And a reboot will let you fall back to the other partition (depending on how you configured it).

What bootloader do you use?

philippbachmann08, 2019-01-29:
Ah so the `boot status` represents if there are any boot attempts left. I use Barebox with a default number of attempts of 3, so the behavior is correct. Therefore I have to explicitly set it to bad manually and check the remaining attempts via `barebox-state -d`. I'm still getting used to the concept, thanks for the clarification!

closed 2019-01-29 by philippbachmann08