rauc/rauc #407: 'rauc status' returns Warning (GDBus error) if all rootfs slots have 0 attempts remaining.

FabianKnapp, 2019-02-14:
However, it is a valid case that all rootfs slots have 0 attempts remaining, since we have barebox configured to reset remaining attempts on power-on.

This is the complete output of 'rauc status':

(rauc:453): rauc-WARNING **: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code30: Failed getting primary slot: Failed getting primary slot: Unable to obtain primary element

```
Compatible:  abc
Variant:     
Booted from: system1
Activated:   (null) ((null))
slot states:
  dtb.1: class=dtb, device=/dev/ubi0_5, type=ubivol, bootname=(null)
      state=active, description=, parent=rootfs.1, mountpoint=(none)
  rootfs.0: class=rootfs, device=/dev/ubi0_2, type=ubifs, bootname=system0
      state=inactive, description=, parent=(none), mountpoint=(none)
      boot status=bad
  kernel.1: class=kernel, device=/dev/ubi0_4, type=ubivol, bootname=(null)
      state=active, description=, parent=rootfs.1, mountpoint=(none)
  rootfs.1: class=rootfs, device=/dev/ubi0_6, type=ubifs, bootname=system1
      state=booted, description=, parent=(none), mountpoint=(none)
      boot status=bad
  kernel.0: class=kernel, device=/dev/ubi0_0, type=ubivol, bootname=(null)
      state=inactive, description=, parent=rootfs.0, mountpoint=(none)
  dtb.0: class=dtb, device=/dev/ubi0_1, type=ubivol, bootname=(null)
      state=inactive, description=, parent=rootfs.0, mountpoint=(none)

```

ejoerns, 2019-02-15:
By current definition, if a slot has 0 remaining attempts we consider it being disabled.
Thus when having two slots where the priority is > 0 but their attempts is 0 RAUC sees them as disabled.
Also barebox bootchooser assumes a slot to be disabled in case of 0 remaining attempts. Then it sets it priority to 0.

Normally, you should only hit this case if all your attempts counter became 0. Thus if you have an initial attempt value being larger you should not run into this case. If you start with 1 while having one slot deactivated already, this will surely hit you.

Which is the actual case that led to this scenario? The reset on 'power-on' should only be required in case of a major failure of your setup. Otherwise either the bootloader or the system should reset the attempts counters in case of successful boots. Where do you reset?

The only alternative to the current handling I could guess is using only the slots priority value for primary slot detection. But I am not sure about other drawbacks, yet.

FabianKnapp, 2019-02-28:
> Thus if you have an initial attempt value being larger you should not run into this case. If you start with 1 while having one slot deactivated already, this will surely hit you.

Yep, thats the case. So both slots should contain a valid (identical?) image during production?

ejoerns, 2019-06-04:
@FabianKnapp Error handling is done already according to my understanding. Did you mean something like I have in #443 now?

ejoerns, 2019-10-06:
As there was no further feedback I'm going to close this. From my understanding, the handling is ok so far. @FabianKnapp feel free to reopen if things appear to be still unresolved to you.

We do have error handling for the case mentioned, it is in the service telling that we cannot reliably determine an activated slot. The 'D-Bus' error, in turn, was only caused by missing handling for remote errors which was fixed in #443 as mentioned above.

closed 2019-10-06 by ejoerns