rauc/rauc #412: Get version of current system

jonaskgandersson, 2019-02-20:
Hi
How can I get the version of the running system?
I would like to be able to compare current system with available bundle version, do I have to use some custom script to write the version to some file?

I have looked at [11.3. Slot Status](https://rauc.readthedocs.io/en/latest/reference.html?highlight=version#slot-status)  but cant get it to work, I have tried to add statusfile to my system.conf file but no file is generated.

The optimal for would have been if I can get the version over D-bus.

I use meta-rauc and Yocto.

ejoerns, 2019-02-21:
Hi, could you show the relevant section of you system.conf? Does the RAUC service produce any warnings about not being able to load status file?

If the status file is not gernerated at a global path, you still should have one in each slot containing a writable fs that you updated. There it is located at ``/slot.raucs``

What is your motivation for comparing the version? Note that the 'version' is only a free-text human readable string and not internally evaluated by RAUC.

jonaskgandersson, 2019-02-22:
My attempt:
```
[system]
compatible=device1
bootloader=barebox
mountprefix=/mnt/rauc
statusfile=/data/rauc/status
```
I have not noticed any warning from RAUC, but I haven't looked for them either, are there any log file I can check?

I have search for an slot.raucs file in my entire system but cant find one, neither in an build witch the "statusfile" defined or not. My output from ``rauc status --detailed`` is the following:
```
Compatible:  device1
Variant:     
Booted from: system1
Activated:   rootfs.1 (system1)
slot states:
  rootfs.0: class=rootfs, device=/dev/ubi0_3, type=ubivol, bootname=system0
      state=inactive, description=, parent=(none), mountpoint=(none)
      boot status=good
      slot status:
          bundle:
              compatible=(null)
  rootfs.1: class=rootfs, device=/dev/ubi0_4, type=ubivol, bootname=system1
      state=booted, description=, parent=(none), mountpoint=(none)
      boot status=good
      slot status:
          bundle:
              compatible=(null)
```
I guess that "slot status" maybe lacking som information because of missing status file?

As for now the intention was to compare the current version with an version stored in an host to see if there is an newer version available for download. Our intent is to try out casync and I guess that it have an more elaborate way of determine the versions.

ejoerns, 2019-02-22:
Ok, for the /slot.raucs: when setting `type=ubivol` RAUC cannot make any assumption on the file system actually used on it (could be squashfs or whatever). If you use `type=ubifs` instead, it should work.

Which version of RAUC ist that?

It should note something like `load_slot_status_globally: {error message}.` in case of a failure.
With debug level on `rauc service -d`, it should addtionally note `Load status for slot {slotname}`.

casync will not determine any version for you, as it only maintiains hash indexes. But it will only notice that it can copy the entire system from your current for updating ;)

jonaskgandersson, 2019-02-22:
RAUC version ``rauc 1.0``

I have changed to ``type=ubifs`` but still the same problem:
```
root@device:~ rauc status --detailed
Compatible:  device1
Variant:     
Booted from: system0
Activated:   rootfs.0 (system0)
slot states:
  rootfs.0: class=rootfs, device=/dev/ubi0_3, type=ubifs, bootname=system0
      state=booted, description=, parent=(none), mountpoint=(none)
      boot status=good
      slot status:
          bundle:
              compatible=(null)
  rootfs.1: class=rootfs, device=/dev/ubi0_4, type=ubifs, bootname=system1
      state=inactive, description=, parent=(none), mountpoint=(none)
      boot status=good
      slot status:
          bundle:
              compatible=(null)
```
I get the following error for rauc.service:
```
Feb 22 12:15:23 rauc[408]: mounting slot /dev/ubi0_3
Feb 22 12:15:23 rauc[408]: Failed to load status file /run/rauc/rootfs.0/slot.raucs: No such file or directory
Feb 22 12:15:23 rauc[408]: mounting slot /dev/ubi0_4
Feb 22 12:15:24 rauc[408]: Failed to load status file /run/rauc/rootfs.1/slot.raucs: No such file or directory
```

ejoerns, 2019-02-22:
Note that you must have performed an installation before it is written the first time.

jonaskgandersson, 2019-02-22:
Problem solved!
The last was only because my Yocto build not rebuild the bundle fully.
Thanks!

closed 2019-02-22 by jonaskgandersson

anthonio9, 2024-08-16:
Hi, I've run into a similar problem, but my setup is with `type=ext4`. Is there a way to retrieve the information about the running firmware version in such case? At the moment far the output of `rauc status --detailed` is as below:
```sh
root@imx:~# rauc status --detailed
=== System Info ===
Compatible:  IMX93_QSB
Variant:     
Booted from: rootfs.1 (B)

=== Bootloader ===
Activated: rootfs.1 (B)

=== Slot States ===
x [rootfs.1] (/dev/mmcblk0p3, ext4, booted)
        bootname: B
        mounted: /
        boot status: good
      slot status:
          bundle:
              compatible=(null)
          installed:

o [rootfs.0] (/dev/mmcblk0p2, ext4, inactive)
        bootname: A
        boot status: good
      slot status:
          bundle:
              compatible=(null)
          installed:
```