rauc/rauc #416: calling `rauc status --output-format=json` from install hook fails

mbessler, 2019-02-26:
I'm using a custom install hook, and so far (in rauc 0.4) I had been using the `rauc status` json output inside my hook without issues. But after switching to 1.0, this appears to no longer work. 
```
rauc status --output-format=json | jq -r '[.slots[][] | select(.class | contains("rootfs")) ] | unique_by(.device) | .[] |  .device' 
```
yields this in the log:
```
GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code30: already processing a different method
```
It looks like this may be due to #299. Am I not supposed to use `rauc status` during the install operation?

ejoerns, 2019-02-26:
@mbessler thanks for reporting this!

If I get it right, this is triggered by the check in https://github.com/rauc/rauc/blob/master/src/service.c#L323 which appears to be not valid to me as we do not access the context writing.

mbessler, 2019-02-27:
@ejoerns yes, commenting out the context busy check where you indicated does resolve my issue.

If that does not have negative side-effects, I can create a PR, let me know...

ejoerns, 2019-02-27:
@mbessler current implementation has a negative side-effect ;) Currently, I don't see any, thus feel free to create a patch and we check if tests stay green etc.

mbessler, 2019-02-28:
PR opened: #417

ejoerns, 2019-03-11:
@mbessler We just discussed this and came to the conclusion that it is not safe to call 'rauc status' while an installation is running.
The relevant change was the rework of status handling to be fully D-Bus-based by default. Prior to this, calling 'rauc status' resulted in collection status information from the client process itself. Now it is only some D-Bus calls into the RAUC service. There we cannot assure (because of having no locking) that you will always get a consistent state of status information.

However, calling RAUC from the hook itself should not be required as you are already in a running RAUC instance that has all information available. Thus my question would be why you need to call `rauc status` and obtain slot information? Note that almost all information relevant for the slot currently handled is already available: https://rauc.readthedocs.io/en/latest/using.html#slot-hooks

mbessler, 2019-03-11:
Hi @ejoerns 
In my case the variables provided to the hook aren't sufficient as I need access to the slot names and devices of the other rootfs slots. 

I'm using a scheme somewhat similar to casync, just with a much smaller tool I wrote. Basically casync but limited to images only, and updates in place chunk-by-chunk instead of downloading to temp. storage. The tool lives here: https://github.com/mbessler/imidj.

Like casync, my tool can also use existing images as chunk sources so less data has to be downloaded, so my hook needs to know the device and bootname of other slots in the same class.

ejoerns, 2019-03-12:
@mbessler haven't hat casync in my hand for a while but I assumed it to suppot streaming of data, too. Is that not supported by casync?

Ok, then I get your requirement. I fear, we have the code for this full status information already, but it's not prepared for using this information from a hook but only from the full custom install handler https://rauc.readthedocs.io/en/latest/using.html#full-custom-update

Maybe this is too much information for an install hook, but having a reference to the counter part of the currently updated slot could be a common use case for implementing custom install hooks?!

mbessler, 2019-03-12:
@ejoerns Not according to what I remember looking at in code and behavior, but its been already a few months. On my target I saw casync error out *a lot*, and since its a slower platform, testing was slow and painful, having to run for 10-20min, then erroring out, then try again. Same happened occasionally on my build machine. So I gave up on casync, and eventually wrote a prototype of what would become `imidj` in a day in python.

I looked at custom handlers vs hooks before I started on my hook, and IIRC, the hook allowed the script to focus on a smaller subset of work. But maybe a custom handler would be better... looking into that, thanks. 
Alternatively, maybe creating a handler in C might be the way to go since I've worked out most of the kinks out of my hook. Not sure if that would be something you'd consider upstreaming into rauc proper?

ejoerns, 2019-03-13:
@mbessler I ran into some issues, too when I tried it initially (and was probably one of the first). A few I fixed, but in the meantime and especially in the last months there was a lot of movement in the project thus I would expect it to be more stable now (and I expect to see a v3 release some day.. :D )

I would also see a hook as a more appropriate solution, too, as there is no need to bypass the slot 
selection etc. Thus providing information no the inactive counterpart of the slot sounds reasonable to me. If one wants to expose the entire slot status in a hook too I'm not quite sure.

You mean a C handler specifically for imidj as we have one for casync? In general we might want to accept it but it should add a real benefit and be a project not only used by a single developer ;) Thus if it would turn out that both is valid, we should consider this in the future. For now I find an install hook (or using casync) more appropriate.

jluebbe, 2023-10-24:
Calling `rauc status` from an install hook won't be supported. Instead, if more information than what is available to hooks is needed, native integration into RAUC would be the way to go.

In the meantime, we have adaptive updates (https://rauc.readthedocs.io/en/stable/advanced.html#adaptive-updates). From the description of imidj, it sounds like it could work by storing the `chidx` and `.chblo.xz` in the bundle and accessing the on-demand via streaming.

I'm going to close this issue, as we don't intend to support calling `rauc status` from a hook (which was the original topic of this issue). Feel free to open new issues/PRs for the other discussed aspects.

closed 2023-10-24 by jluebbe

mbessler, 2023-10-24:
I ended up solving this by reading /etc/rauc/system.conf from my hook bash script and parsing with `sed` and `grep`.