rauc/rauc #418: Erase barebox environment

vishwanathchandapur, 2019-03-07:
We have board with emmc and barebox  bootloader, We are trying to experiment with ruac barebox update .while testing we found rauc doesn't clear barebox environment after upgrade because which next boot old barebox environment will be used.

jluebbe, 2019-03-07:
I'd recommend avoiding a barebox environment for any non-development use-case. You can store dynamic data with a fixed schema using the state framework: https://barebox.org/doc/latest/user/state.html.

ejoerns, 2019-03-07:
@vishwanathchandapur there is nothing wrong with that, that is what one would expect. The environment should be persistent as it might contain changes unrelated to the specific barebox you use.

But this should not be an issue as you should not have anything in you barebox env on a production system. The barebox env is only meant for developing or debugging purpose!

When you update barebox on eMMC, do you use the boot partitions for that?

ejoerns, 2019-03-07:
Ah, that happens if no reloading the browser before finally submitting a pre-written answer... :D

vishwanathchandapur, 2019-03-07:
Yes as of now  we are using boot partitions and we trying experiment  with barebox update on filed if this doesn't have any side effects .
Coming to your solution in scenarios where if try to pass new argument in linux boot arguments ,the new changes wont be reflected untill we clear barebox environment?

ejoerns, 2019-03-21:
If using barebox I would recommend using [bootspec](https://www.freedesktop.org/wiki/Specifications/BootLoaderSpec/) files for adding linux boot arguments so that they are part of the rootfs instead of placing them in the barebox env.

> Yes as of now we are using boot partitions and we trying experiment with barebox update on filed if doesn't have any side effects .

I did not get what you actually mean by barebox update on file{d,s}. Could you rephrase that?

ejoerns, 2019-06-04:
@vishwanathchandapur Any updates on this? Does it work for you now?

ejoerns, 2019-08-15:
I will close this now as there was no feedback for 5 months and there does not seem to be any general issue here.

closed 2019-08-15 by ejoerns