rauc/rauc #419: combo slot-group bundles are not supported

leiflm, 2019-03-07:
**Problem**: [Currently](https://github.com/rauc/rauc/releases/tag/v1.0) RAUC rejects a provided bundle, if it cannot [determine a suitable target slot](https://rauc.readthedocs.io/en/latest/basic.html#target-slot-selection) for every provided image. This is by intention to prevent unintended mistakes.
However, it prevents one from providing updates for all slots-groups of a system with a _single_ bundle. Especially in the case of an [asymmetric system](https://rauc.readthedocs.io/en/latest/scenarios.html#asymmetric-slots) it might be desirable to provide a complete update (i.e. `main` _and_ `update`).

**Use-Case**: Update multiple slots, one after the other, using a single file/bundle.

leiflm, 2019-03-07:
a mechanism, similar to the [variant-mechanism](https://rauc.readthedocs.io/en/v1.0/advanced.html#handling-board-variants-with-a-single-bundle), informing RAUC to eventually ignore certain images of a provided bundle, instead of rejecting the bundle entirely could be implemented.

leiflm, 2019-03-18:
to be clear: I'd provide an implementation of the [suggested method](https://github.com/rauc/rauc/issues/419#issuecomment-470520108) **if** upstream approves on the general idea/way.

ejoerns, 2019-03-21:
We discussed this last week. I think what the use case description misses is the reason why one need a single bundle for this.
The normal use case for A/recovery would be to only update A and update recovery only when this is really required. In this case a dedicated bundle is more appropriate.
The other use case I see could be to keep A and recovery in sync. Then one can place the recovery bundle inside the rootfs of a bundle as one would always first update A and then the recovery system.

We are willing to accept solutions for potentially common use cases, but would like to prevent from implementing too many 'special solutions' that potentially affect stability in the future.

leiflm, 2019-03-21:
> The other use case I see could be to keep A and recovery in sync. Then one can place the recovery bundle inside the rootfs of a bundle as one would always first update A and then the recovery system.

That's precisely the use-case here: keeping systems in sync. For the suggested work-around however, one would need extra space on A. But that extra space is used to provide a fallback bundle stored within the recovery system.