rauc/rauc #426: Increasing memory usage of rauc in idle state

lategoodbye, 2019-04-05:
Hi, we are using rauc ( https://github.com/rauc/rauc/commit/cde7b1a2f8699b2fd11bbf556d93fbff85ea3884 ) on a ARMv7 platform with Debian Stretch. I noticed a increasing memory usage of rauc even in idle state. After boot `ps aux` gives me RSS 9676 and after 17 hours i'm getting RSS 9932.

I installed valgrind (using the bugfixed version from buster) on the platform and give it a short idle run (~ 2 min):

```
==1583== LEAK SUMMARY:
==1583==    definitely lost: 66 bytes in 9 blocks
==1583==    indirectly lost: 153 bytes in 10 blocks
==1583==      possibly lost: 1,320 bytes in 26 blocks
==1583==    still reachable: 233,112 bytes in 7,776 blocks
==1583==                       of which reachable via heuristic:
==1583==                         newarray           : 1,384 bytes in 46 blocks
==1583==         suppressed: 0 bytes in 0 blocks
```

I attached the full output.

glib2-version: 2.50.3-2

[rauc-valgrind.txt](https://github.com/rauc/rauc/files/3047559/rauc-valgrind.txt)

ejoerns, 2019-04-05:
Hi @lategoodbye , could you please verify if the issue you encountered is fixed by https://github.com/rauc/rauc/pull/383 ?

I'd suggest to use at least RAUC 1.0 anyway.

lategoodbye, 2019-04-12:
@ejoerns I updated to RAUC 1.0 and the leak(s) still per persists. My first version of the valgrind output wasn't correct parameterized for glib here is a better version.

[rauc-1.0-valgrind.log](https://github.com/rauc/rauc/files/3072210/rauc-1.0-valgrind.log)

lategoodbye, 2019-04-13:
Looking at network.c shows a call to `curl_global_init` but there is no matching call to `curl_global_cleanup`. Also the GOptionGroup handling in main.c looks fishy.
Sure this cannot be responsible for an increasing memory usage, but this all together makes it harder to identify the real problem.

ejoerns, 2019-04-15:
@lategoodbye Thanks for further investigating the issue(s). I agree, the `curl_global_cleanup()` is missing. That is mainly because the network code was only rarely touched before we started supporting casync. Do you use network capabilities of RAUC?
The `GOptionGroup` handling surely lacks some cleanups, but that should not lead to increased memorey consumption in the service. Anyway, should be fixed.

Would you like to make a PR for the issues you revealed?

Maybe also check the just opened PR #428 becuase this looks like a good candidate for increasing leakage.

lategoodbye, 2019-04-15:
@ejoerns We compiled RAUC with network support, but actually don't use it.

Sorry, i'm not familiar with glibc and i couldn't find a proper solution because the context stuff looks very tricky to me. So someone else should take care of these.

Sure i will look at #428 and give feedback.

ejoerns, 2019-04-15:
@lategoodbye no problem, thanks for the feedback. I will check the cleanup handling as soon as I find  some free minutes

lategoodbye, 2019-04-24:
@ejoerns I tested #428 with a long-term test over the holidays and the RSS stayed the same. So this seems to fix the issue.

ejoerns, 2019-04-24:
@lategoodbye Yeah, that's good news! Thanks for testing! :+1:  And @mhei thanks for fixing!

I assume we can close this then. Feel free to reopen if required.

closed 2019-04-24 by ejoerns