rauc/rauc #427: Install archive on already mounted slot.

gajanak, 2019-04-07:
Hi,

I need a simple solution for update a single ext4 partition which is already mounted.
I know, documentation recommend to boot a rescue system, update and reboot again.

But here I would present a additional optional way.

In #[Commit](https://github.com/gajanak/rauc/commit/45bc88a5c1233a4e4f676461d7f7c057d7cde57a) you can see my solution, adding net Option 'install-on-mounted' will allow to switch to different update handler , if destination slot is already mounted.

Would you prefer a complete another solution ? 
Any comment would be nice to know.

ejoerns, 2019-04-15:
Linking the other open discussion threads here for reference, first of all: #186 #326

gajanak, 2019-04-15:
Oh, I missed this Issues on searching. Soory. I will study it ...
Can you look at my diff, is it a possible solution ?

ejoerns, 2020-10-25:
@gajanak In the meantime and after some discussion, #326 got merged and #186 rejected.

Are the changes you said you could split out in a separate PR still relevant? If so, feel free to make one, if not, please let me know.

I still do not see the handling of already-mounted slots as being fully resolved. What we have in #326 is the absolute minimum we should support to give users some degree of freedom.

However, in two of my projects I am currently working on I face similar issues and we decided, for now, to simply drop the mount of the /boot slot. This was possible in both cases as the only thing that needed to be accessed from there would have been the bootloader environment and this conflicted anyway with the requirement to allow atomic updates for the bootloader.