rauc/rauc #431: compatible check multiple entries

hpatriarche, 2019-05-02:
According to code verify_compatible, it is not possible to specifiy several compatible value in a bundle and if one is equal to system.conf on the target then install it.
May I am wrong ?

I mean it would be great to tell that a bundle is compatible with several HW.

Thanks, Herve.

ejoerns, 2019-05-09:
We thought about this but decided to 'keep it simple'.
If a bundle needs to cover multiple compatibles, maybe simply the scope of the compatibles is too narrow?

We have the dedicated mechanism of 'variants' that aims to handle partial variations in hardware: https://rauc.readthedocs.io/en/latest/advanced.html#handling-board-variants-with-a-single-bundle
Maybe this could also be what you require?

Regards, Enrico

ejoerns, 2019-06-04:
@hpatriarche Do you have any further opinions on this concerning my comments?

closed 2019-10-06 by ejoerns

ejoerns, 2021-10-08:
> My use case for this would be updating the compatible string.
> 
> In the past we used a generic compatible string like `project`. With changing requirements we need a seperate variant for North-America (`project-northamerica`) now. With that I'd like to rename the old `project` to `project-europe`. Right now I'd need an intermediate image to update the compatible string in the `system.conf` on the boards. If a bundle could be compatible to both `project` and `project-europe` that would not be necessary and keeps the possibility for the customer to skip versions.

Taking the approach that you support two compatibles and would like to skip updates. Then customer A installs the update containing two compatibles and customer B doesn't. Then, in the following update, you make your bundle compatible with new 'project-europe' only. Then customer A is able to install it while customer B is not. Thus this only shifts the problem and maybe makes the migration phase a bit softer. But it does not solve the problem that you would need to know when all customers have updated to the new compatible, doesn't it?

Because of these issues, we have "System compatible is specific enough" in the [checklist](https://rauc.readthedocs.io/en/latest/checklist.html) as this could always cause problems in the future.

Anyway, for me, the straight-forward solution for what you described would be to stay with `project` for the europe variant and give all further variants more specific names (like `project-northamerica`). Wouldn't that be an option?

> RUAC variants don't quite cut it as the difference is in the rootfs and therefore the images would be quite large. Also the user should not be able flash `project-europe` on `project-northamerica` and vice versa.

Note that compatible strings are only a safety feature and should not be used for enforcing bundle installation policies. This is what cryptography is for and you can resolve this by e.g. using different key sets for signing the europe and america version.

> Note: Our project is not yet launched so I can totally live with an intermediate image. I just wanted to share a possible use case for multiple compatibiliy strings in a bundle.

Thanks for your input :+1: