rauc/rauc #437: Failed updating slot rootfs.1: opening output device failed: Device or resource busy

serra82, 2019-05-26:
I'm integrating the rauc recipies, but the upgrade process fails.

When I try to manually upgrade the image:
rauc install /tmp/allwinner-rauc-orange-pi-zero-plus.raucb

the process fails:
rauc-Message: 17:23:07.522: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: installing
rauc-Message: 17:23:07.523: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:   0% Installing
rauc-Message: 17:23:07.524: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:   0% Determining slotstates
rauc-Message: 17:23:07.527: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  20% Determining slot states done.
rauc-Message: 17:23:07.527: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  20% Checking bundle
rauc-Message: 17:23:07.534: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  20% Verifying signature
rauc-Message: 17:23:07.750: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  40% Verifying signature done.
rauc-Message: 17:23:07.767: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  40% Checking bundle done.
rauc-Message: 17:23:07.798: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  40% Loading manifest file
rauc-Message: 17:23:07.801: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  60% Loading manifest file done.
rauc-Message: 17:23:07.801: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  60% Determining target install group
rauc-Message: 17:23:07.802: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  80% Determining target install group done.
rauc-Message: 17:23:07.867: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  80% Updating slots
rauc-Message: 17:23:19.624: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  80% Checking slot rootfs.1
rauc-Message: 17:23:19.628: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  90% Checking slot rootfs.1 done.
rauc-Message: 17:23:19.628: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb:  90% Copying image to rootfs.1
rauc-Message: 17:23:19.628: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: 100% Copying image to rootfs.1 failed.
rauc-Message: 17:23:19.629: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: 100% Updating slots failed.
rauc-Message: 17:23:19.945: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: 100% Installing failed.
rauc-Message: 17:23:19.999: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: LastError: Installation error: Failed updating slot rootfs.1: opening output device failed: Device or resource busy
Installing `/tmp/allwinner-rauc-orange-pi-zero-plus.raucb` failed

journalctl -u rauc

May 25 13:43:58 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: finished[[0m 
May 25 13:43:58 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39minstalling `/tmp/allwinner-rauc-orange-pi-zero-plus.raucb` failed: 1[[0m
May 25 16:28:35 orange-pi-zero-plus rauc[359]: input bundle: /tmp/allwinner-rauc-orange-pi-zero-plus.raucb
May 25 16:28:35 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mActive slot bootname: /dev/mmcblk0p2[[0m
May 25 16:28:35 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: started[[0m
May 25 16:28:35 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: Checking and mounting bundle...[[0m
May 25 16:28:35 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mReading bundle: /tmp/allwinner-rauc-orange-pi-zero-plus.raucb[[0m
May 25 16:28:35 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mVerifying bundle... [[0m
May 25 16:28:35 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mMounting bundle '/tmp/allwinner-rauc-orange-pi-zero-plus.raucb' to '/run/rauc/bundle'[[0m
May 25 16:28:35 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mMarking target slot rootfs.1 as non-bootable...[[0m
May 25 16:28:35 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: Updating slots...[[0m
May 25 16:28:49 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mChecking image type for slot type: ext4[[0m
May 25 16:28:49 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mImage detected as type: *.ext4[[0m
May 25 16:28:49 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: Checking slot rootfs.1[[0m
May 25 16:28:49 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mSlot needs to be updated with /run/rauc/bundle/allwinner-image-orange-pi-zero-plus.ext4[[0m
May 25 16:28:49 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mUpdating /dev/mmcblk0p3 with /run/rauc/bundle/allwinner-image-orange-pi-zero-plus.ext4[[0m
May 25 16:28:49 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mopening slot device /dev/mmcblk0p3[[0m
May 25 16:28:49 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: Updating slot rootfs.1[[0m
May 25 16:28:50 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39mInstallation error: Failed updating slot rootfs.1: opening output device failed: Device or resource busy[[0m
May 25 16:28:50 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: Installation error: Failed updating slot rootfs.1: opening output device failed: Device or resource busy[[0m
May 25 16:28:50 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39minstalling /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: finished[[0m
May 25 16:28:50 orange-pi-zero-plus rauc[359]: [[0;1;39m[[0;1;31m[[0;1;39minstalling `/tmp/allwinner-rauc-orange-pi-zero-plus.raucb` failed: 1[[0m

I can't understand why: 
Installation error: Failed updating slot rootfs.1: opening output device failed: Device or resource busy
/dev/mmcblk0p3 isn't mounted;
if I run:
rauc status mark-active rootfs.1
slot.raucs file is correctly created in /dev/mmcblk0p3

Any idea how to solve this problem?
Thank you.

ejoerns, 2019-05-28:
Hi @serra82,

it seems that someone else is using /dev/mmcblk0p3. Does

    fuser /dev/mmcblk0p3
    fuser /dev/mmcblk0p3p1
    lsof | grep mmcblk0p3

show any process using the device?

serra82, 2019-05-28:
Hi @ejoerns,

nobody uses the device:
root:~# fuser /dev/mmcblk0p3
root:~# fuser /dev/mmcblk0p3p1
fuser: can't stat '/dev/mmcblk0p3p1': No such file or directory
root:~# lsof | grep mmcblk0p3

if can be usefuel:
root:~# dmesg |grep mmcblk0
[    1.514742] mmcblk0: mmc0:59b4 00000 14.9 GiB
[    1.516041]  mmcblk0: p1 p2 p3 p4
[    1.841849] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    3.666627] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    8.114457] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)

You can see my commit with rauc integration:
https://bitbucket.org/serra82/meta-allwinner-hx/commits/35968925e7b12840704088c502ab89873b0daa13

thank you

ejoerns, 2019-05-28:
Ouh, typo in my example I meant

    fuser /dev/mmcblk0

instead of

    fuser /dev/mmcblk0p3p1

The dmesg does not really help without knowing when this happened. Thus maybe you post the relevant (from the time when the update occured) output of

    journalctl --no-pager

that includes both what happened in RAUC as well as in the kernel etc.

serra82, 2019-05-28:
I don't see nothing of relevant
```
May 28 16:13:53 orange-pi-zero-plus rauc[360]: input bundle: /tmp/allwinner-rauc-orange-pi-zero-plus.raucb
May 28 16:13:53 orange-pi-zero-plus rauc[360]: Active slot bootname: /dev/mmcblk0p2
May 28 16:13:53 orange-pi-zero-plus rauc[360]: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: started
May 28 16:13:53 orange-pi-zero-plus rauc[360]: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: Checking and mounting bundle...
May 28 16:13:53 orange-pi-zero-plus rauc[360]: Reading bundle: /tmp/allwinner-rauc-orange-pi-zero-plus.raucb
May 28 16:13:53 orange-pi-zero-plus rauc[360]: Verifying bundle...
May 28 16:13:53 orange-pi-zero-plus rauc[360]: Mounting bundle '/tmp/allwinner-rauc-orange-pi-zero-plus.raucb' to '/run/rauc/bundle'
May 28 16:13:53 orange-pi-zero-plus rauc[360]: Marking target slot rootfs.1 as non-bootable...
May 28 16:13:53 orange-pi-zero-plus rauc[360]: ## Error: "BOOT_ORDER" not defined
May 28 16:13:53 orange-pi-zero-plus rauc[360]: Unable to obtain BOOT_ORDER: fw_printenv failed with exit code: 1
May 28 16:13:53 orange-pi-zero-plus rauc[360]: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: Updating slots...
May 28 16:14:03 orange-pi-zero-plus rauc[360]: Checking image type for slot type: ext4
May 28 16:14:03 orange-pi-zero-plus rauc[360]: Image detected as type: *.ext4
May 28 16:14:03 orange-pi-zero-plus rauc[360]: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: Checking slot rootfs.1
May 28 16:14:03 orange-pi-zero-plus rauc[360]: Slot needs to be updated with /run/rauc/bundle/allwinner-image-orange-pi-zero-plus.ext4
May 28 16:14:03 orange-pi-zero-plus rauc[360]: Updating /dev/mmcblk0p3 with /run/rauc/bundle/allwinner-image-orange-pi-zero-plus.ext4
May 28 16:14:03 orange-pi-zero-plus rauc[360]: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: Updating slot rootfs.1
May 28 16:14:03 orange-pi-zero-plus rauc[360]: opening slot device /dev/mmcblk0p3
May 28 16:14:03 orange-pi-zero-plus rauc[360]: Installation error: Failed updating slot rootfs.1: opening output device failed: Device or resource busy
May 28 16:14:03 orange-pi-zero-plus rauc[360]: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: Installation error: Failed updating slot rootfs.1: opening output device failed: Device or resource busy
May 28 16:14:03 orange-pi-zero-plus rauc[360]: installing /tmp/allwinner-rauc-orange-pi-zero-plus.raucb: finished
May 28 16:14:03 orange-pi-zero-plus rauc[360]: installing `/tmp/allwinner-rauc-orange-pi-zero-plus.raucb` failed: 1
```
I can't understand why rauc see mmcblk0p3 busy  :-(
```
root:~# fuser /dev/mmcblk0
```

ejoerns, 2019-06-04:
Unfortunately, I don't have a good idea. Does this only happen for p3 or also when updating p2?

Did you proceed with analyzing what is going wrong?

bendelec, 2019-08-01:
Hi serra82,

were you able to solve your problem? If yes, could you please take a moment to tell us how?

I seem to have the same issue on a digi ccimx8x based  board. When I have booted from rootfs.0 (mmcblk0p3), any attempt to update rootfs.1 (mmcblk0p4) fails with "Installation error: Failed updating slot rootfs.1: opening output device failed: Device or resource busy" (from journalctl output).
- /dev/mmcblk0p3 is not mounted. manually mounting and unmounting it doesn't solve the issue
- lsof and fuser show nothing

thanks for your status update ;)

serra82, 2019-08-01:
Hi,
@ejoerns also p2
@bendelec I didn't resolve the problem yet

ejoerns, 2019-11-07:
@serra82 @bendelec I would like to further investigate this if it apparently occurs in multiple scenarios.

The one thing I see both cases have in common is that it happens on SD/eMMC. Unsure if that is just correlation or causality.

If one of you (or both) have the option to provide the *full* journal during the time of updating and/or start the RAUC service with `strace` and providing the output, that could be quite helpful to trace down the issue.

I expect either something in the system to shortly access the device (maybe triggered by RAUC) or a race in RAUC's handling itself.

Additionally, it would be great if you can check that the recent (1.2) version still shows this behavior
Thanks!

ejoerns, 2020-03-04:
@serra82 @bendelec Any new hints/logs for this that could help to debug?

serra82, 2020-03-04:
Hi @ejoerns, sorry but actually we aren't working to rauc

ejoerns, 2020-10-24:
I'll close this now as I cannot reproduce the issue and no one who could currently seems to be using RAUC or facing this issue which makes debugging impossible.

Anyone facing this issue: feel free to reopen.

closed 2020-10-24 by ejoerns

picostef, 2023-02-28:
Hello, 
I faced the same issue with rauc version 1.8. I could not solve it but I think I found the source of the issue, so here are my 2 cents on this issue!

The partition I was trying to update was an extended partition, trying to write on another extended partition:
`
Device         Boot   Start     End Sectors  Size Id Type
/dev/mmcblk0p1         2048   12287   10240    5M 83 Linux
/dev/mmcblk0p2        12288  114687  102400   50M 83 Linux
/dev/mmcblk0p3       114688  638975  524288  256M 83 Linux
/dev/mmcblk0p4       638976 7634943 6995968  3.3G  5 Extended
/dev/mmcblk0p5       641024  743423  102400   50M 83 Linux
/dev/mmcblk0p6       745472 3891199 3145728  1.5G 83 Linux
/dev/mmcblk0p7      3893248 7430143 3536896  1.7G 83 Linux
/dev/mmcblk0p8      7432192 7634943  202752   99M 83 Linux
`

when I update /dev/mmcblk0p2 with  rauc from /dev/mmcblk0p6, or the reverse, no problem.
when I update /dev/mmcblk0p8 with  rauc from /dev/mmcblk0p6, I get a device busy error. I suspect it is due to /dev/mmcblk0p4 that is somehow considered open...

ejoerns, 2023-02-28:
@picostef  Do you have `mount` output and the rauc service log of the installation causing the error?

picostef, 2023-02-28:
sure:
`/dev/mmcblk0p6 on / type ext4 (ro,noatime,nodioread_nolock,nodelalloc,data=journal)
devtmpfs on /dev type devtmpfs (rw,relatime,size=854816k,nr_inodes=213704,mode=755)
proc on /proc type proc (rw,relatime)
/dev/mmcblk0p7 on /opt/data type ext4 (rw,noatime,sync,nodioread_nolock,nodelalloc,data=journal)
/dev/mmcblk0p8 on /opt/manuf type ext4 (ro,noatime,nodelalloc,data=journal)
/dev/mmcblk2p1 on /mnt/factory type ext4 (rw,noatime,sync,nodioread_nolock,nodelalloc,data=journal)
/dev/mmcblk2p2 on /mnt/update type ext4 (rw,noatime,sync,nodioread_nolock,nodelalloc,data=journal)
tmpfs on /var/volatile type tmpfs (rw,relatime)
tmpfs on /etc/docker type tmpfs (rw,relatime,size=64k)
sysfs on /sys type sysfs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
overlay on /var/lib type overlay (rw,relatime,lowerdir=/var/lib,upperdir=/var/volatile/lib,workdir=/var/volatile/.lib-work)
/dev/mapper/encrypted on /mnt/secure type ext4 (rw,relatime)
cgroup on /sys/fs/cgroup type tmpfs (rw,relatime,mode=755)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu type cgroup (rw,relatime,cpu)
cgroup on /sys/fs/cgroup/cpuacct type cgroup (rw,relatime,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,relatime,blkio)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,relatime,memory)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,relatime,freezer)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,relatime,perf_event)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,relatime,pids)
name=systemd on /sys/fs/cgroup/systemd type cgroup (rw,relatime,name=systemd)
`
before to execute the rauc install I unmount /dev/mmcblk0p8

Engr-Ehsan-Elahi, 2025-02-20:
### LastError: Installation error: Failed updating slot rootfs.1: Failed to open output file/device /dev/mmcblk1p3 failed: Device or resource busy

I am also facing this error while installing raucb in my target. Here are some details which can help you to understand:

``` sh
> rauc install /tmp/update-bundle.raucb 

installing
  0% Installing
E/TA:  check_mechanism_against_processing:140 Processing 0/RSA_PKCS_KEY_PAIR_GEN not permitted (7/0)

 10% Determining slot states done.
 10% Checking bundle
 10% Verifying signature
 20% Verifying signature done.
 20% Checking bundle done.
 20% Checking manifest contents
 30% Checking manifest contents done.
 30% Determining target install group
 40% Determining target install group done.
 40% Updating slots
 40% Checking slot rootfs.1 (B)
 46% Checking slot rootfs.1 (B) done.
 46% Copying image to rootfs.1
 99% Copying image to rootfs.1 failed.
 99% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot rootfs.1: Failed to open output file/device /dev/mmcblk1p3 failed: Device or resource busy
Installing `/tmp/update-bundle.raucb` failed

```
``` sh
> umount /dev/mmcblk1p3                                                                                                                                                     

umount: /dev/mmcblk1p3: not mounted.
```
``` sh
> journalctl -u rauc

Installation e78cf92c failed: Installation error: Failed updating slot rootfs.1: Failed to open output file/device /dev/mmcblk1p3 failed: Device or resource by
Feb 20 11:29:33 rauc[1163]: Installation error: Failed updating slot rootfs.1: Failed to open output file/device /dev/mmcblk1p3 failed: Device or resource busy
```

avmaslovwork-design, 2026-09-17:
Hello, i had the same issue and resolved it. 
Strace showed that rauc fails here 
```
static GUnixOutputStream* open_unix_output_stream(const gchar *filename, int flags, int mode, int *fd, GError **error)
{
	GUnixOutputStream *outstream = NULL;
	int fd_out;

	fd_out = g_open(filename, O_WRONLY | flags, mode);

	if (fd_out == -1) {
		g_set_error(error, R_UPDATE_ERROR, R_UPDATE_ERROR_FAILED,
				"Failed to open output file/device %s failed: %s", filename, strerror(errno));
		return NULL;
	}

	outstream = G_UNIX_OUTPUT_STREAM(g_unix_output_stream_new(fd_out, TRUE));
	if (outstream == NULL) {
		g_set_error(error, R_UPDATE_ERROR, R_UPDATE_ERROR_FAILED,
				"Failed to create stream for output device %s", filename);
		return NULL;
	}

	if (fd != NULL)
		*fd = fd_out;

	return outstream;
}
```
Next i watched that rauc needs O_EXCL flag for open (https://man7.org/linux/man-pages/man2/open.2.html#:~:text=O_EXCL%20Ensure%20that%20this%20call%20creates%20the%20file%3A%20if%20this%20flag%20is) 
Rauc do everything correct : it need exclusive flag to avoid update corruption 
Next i used this script to check it 
```
import os

for p in [
    "/dev/mmcblk0p2", # ok
    "/dev/mmcblk0p12", # resource busy
]:
    try:
        fd = os.open(p, os.O_RDWR | os.O_EXCL)
        os.close(fd)
        print(p, "EXCL OK")
    except OSError as e:
        print(p, f"EXCL FAILED errno={e.errno} {e}")
```
In my case something already mounted this. Then i started to watch dmesg log and shoed something like 
```
dmesg | grep mmcblk
[    2.784885] mmcblk0: mmc0:0001 BJTD4R 29.1 GiB 
[    2.788010]  mmcblk0: p1 p2 p3 p4 p5 p6 p7 p8 p9 p10 p11 p12 p13 p14 p15
[    2.789986] mmcblk0boot0: mmc0:0001 BJTD4R 4.00 MiB 
[    2.790786] mmcblk0boot1: mmc0:0001 BJTD4R 4.00 MiB 
[    2.791454] mmcblk0rpmb: mmc0:0001 BJTD4R 4.00 MiB, chardev (236:0)
[    3.555934] EXT4-fs (mmcblk0p12): mounted filesystem with ordered data mode. Quota mode: none.
[    5.613946] EXT4-fs (mmcblk0p12): re-mounted. Quota mode: none.
[    5.649147] EXT4-fs (mmcblk0p14): recovery complete
[    5.649875] EXT4-fs (mmcblk0p14): mounted filesystem with ordered data mode. Quota mode: none.
[    5.799831] EXT4-fs (mmcblk0p15): recovery complete
[    5.800640] EXT4-fs (mmcblk0p15): mounted filesystem with ordered data mode. Quota mode: none.
```
And by timings i found out that was udev rules service in my initramfs. This is indirectly confirmed by the fact that rauc was able to correctly update the boot partition on my device that is not ext4. 
I added mmcblk devicies in ignore list and it works correct 
```
cat udev-extraconf/mount.ignorelist 
/dev/loop
/dev/ram
/dev/mtdblock
/dev/md
/dev/dm-*
/dev/mmcblk0p*
```
After thet rauc works good 
```
rauc install ota-bundleraucb 
installing
  0% Installing
  0% Determining slot states
 10% Determining slot states done.
 10% Checking bundle
 10% Verifying signature
 20% Verifying signature done.
[ 1516.440107] loop0: detected capacity change from 0 to 549450
 20% Checking bundle done.
[ 1516.461734] loop0: detected capacity change from 549450 to 549448
 20% Checking manifest contents
 30% Checking manifest contents done.
 30% Determining target install group
 40% Determining target install group done.
 40% Updating slots
 99% Copying image to rootfs.0 done.
 99% Updating slots done.
100% Installing done.
Installing `/home/root/ota-bundleraucb` succeeded
```
ps. Hope it helps somebody in the future