rauc/rauc #452: Cross-compiling for ARM was recently broken

marv, 2019-06-16:
Cross-compiling RAUC for ARMv7 fails to build with the following errors (actually warnings that got treated as errors due to -Werror):

```
src/mbr.c:26:10: error: packed attribute causes inefficient alignment for 'disk_signature' [-Werror=attributes]
  guint32 disk_signature;
          ^~~~~~~~~~~~~~
src/mbr.c:27:10: error: packed attribute causes inefficient alignment for 'unused' [-Werror=attributes]
  guint16 unused;
          ^~~~~~
src/mbr.c: In function 'is_region_free':
src/mbr.c:164:32: error: format '%lx' expects argument of type 'long unsigned int', but argument 5 has type 'guint64' {aka 'long long unsigned int'} [-Werror=format=]
      "Region start address 0x%lx is in area of "
                              ~~^
                              %llx
src/mbr.c:166:6:
      region_start, i, p_start, p_end);
      ~~~~~~~~~~~~               
src/mbr.c:164:6: error: format '%lx' expects argument of type 'long unsigned int', but argument 7 has type 'guint64' {aka 'long long unsigned int'} [-Werror=format=]
      "Region start address 0x%lx is in area of "
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/mbr.c:166:23:
      region_start, i, p_start, p_end);
                       ~~~~~~~
src/mbr.c:165:25: note: format string is defined here
      "partition %d (0x%lx - 0x%lx)",
                       ~~^
                       %llx
src/mbr.c:164:6: error: format '%lx' expects argument of type 'long unsigned int', but argument 8 has type 'guint64' {aka 'long long unsigned int'} [-Werror=format=]
      "Region start address 0x%lx is in area of "
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/mbr.c:166:32:
      region_start, i, p_start, p_end);
                                ~~~~~
src/mbr.c:165:33: note: format string is defined here
      "partition %d (0x%lx - 0x%lx)",
                               ~~^
                               %llx
src/mbr.c:174:30: error: format '%lx' expects argument of type 'long unsigned int', but argument 5 has type 'guint64' {aka 'long long unsigned int'} [-Werror=format=]
      "Region end address 0x%lx is in area of "
                            ~~^
                            %llx
src/mbr.c:176:6:
      region_start + region_size - 1, i, p_start,
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/mbr.c:174:6: error: format '%lx' expects argument of type 'long unsigned int', but argument 7 has type 'guint64' {aka 'long long unsigned int'} [-Werror=format=]
      "Region end address 0x%lx is in area of "
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/mbr.c:176:41:
      region_start + region_size - 1, i, p_start,
                                         ~~~~~~~
src/mbr.c:175:25: note: format string is defined here
      "partition %d (0x%lx - 0x%lx)",
                       ~~^
                       %llx
src/mbr.c:174:6: error: format '%lx' expects argument of type 'long unsigned int', but argument 8 has type 'guint64' {aka 'long long unsigned int'} [-Werror=format=]
      "Region end address 0x%lx is in area of "
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/mbr.c:177:6:
      p_end);
      ~~~~~
src/mbr.c:175:33: note: format string is defined here
      "partition %d (0x%lx - 0x%lx)",
                               ~~^
                               %llx
src/mbr.c: In function 'r_mbr_switch_clear_partition':
src/mbr.c:356:39: error: format '%lu' expects argument of type 'long unsigned int', but argument 5 has type 'guint64' {aka 'const long long unsigned int'} [-Werror=format=]
     "Failed to set file to position %lu: %s",
                                     ~~^
                                     %llu
     dest_partition->start, g_strerror(errno));
     ~~~~~~~~~~~~~~~~~~~~~              
cc1: all warnings being treated as errors
```

Full log: https://gist.github.com/marv/2b58e91f01c235cecfb8de6102db0503

The format errors would be rather easy to fix using https://developer.gnome.org/glib/stable/glib-Basic-Types.html#G-GUINT64-FORMAT:CAPS and the like, but I'm not sure what to do about the inefficient alignment due to packing

jluebbe, 2019-06-17:
Ah, yes, I'll prepare a fix. Also it seems we need to figure out how to run cross-builds on travis, to avoid this in the future.

ejoerns, 2019-06-18:
Fixed by #453

closed 2019-06-18 by ejoerns