rauc/rauc #457: Options for EXT4 formatting

embeddedrocks, 2019-06-24:
Hi,

is it possible to define options for `mkfs` on EXT4 partitions? Reading through doc and code, it doesn't seem so.

Best Regards

ejoerns, 2019-06-24:
Hi @embeddedrocks,

you're right, it is not supported yet.
But I don't see anything that speaks against implementing this.
Could you point out what your concrete use case for this is?

Best regards, Enrico

Bastian-Krause, 2019-06-24:
Assuming your mkfs.ext4 is a mke2fs symlink, you can configure it via /etc/mke2fs.conf, see `man mke2fs.conf`.

embeddedrocks, 2019-06-24:
Many thanks for your comments!

@ejoerns Use Case is to disable journal for the file system. Implementing it would be another great feature for RAUC (at least for us :-))

@Bastian-Krause Brilliant! Thanks a lot. That should do the trick by writing the file via pre-install hook and than go from there. Cheers mate! 

Best Regards,
Enrico

jluebbe, 2019-06-24:
@embeddedrocks, as options such as journal support are likely to differ between slots, I'd prefer an explicit way to configure it (like `extra-mount-opts=<options>`) in the system conf.
But maybe using the per-device defaults in mke2fs.conf is enough. I would ship it as part of the rootfs, though.

embeddedrocks, 2019-06-25:
@jluebbe many thanks for your further notes and suggestions. I agree, having an explicit way of configuring is preferred. But I would recommend configure it within the update bundle itself. This allows for changes when the next update takes place. The system conf only helps if the new system is already in place. In our case we need a procedure that takes place before the new update is installed and is shipped within the update file itself.
For now we try using the pre-install slot hook before rootfs is written (as you suggested), but we are more than happy to try a new version once that is implemented :-)

Again, many thanks and Best Regards,
Enrico

ejoerns, 2019-07-10:
I agree with @jluebbe that such an option will be useful in some cases and that a global mke2fs.conf will not reflect cases where different slots have different configuration.

Even if @embeddedrocks needs it for migration-purpose I would see this setting more in the system.conf. One always has the option to do a two-step update approach if really required.

@embeddedrocks would you like to implement such an option? Or is your pre-install solution sufficient for now and you don't need it during normal operation?

embeddedrocks, 2019-07-12:
Personally I would prefer a configuration within the update bundle itself, but understand where you're coming from. A per slot configuration is more flexible and you're right @ejoerns that a two-step update would solve it (although this is not best practise from a customers perspective).
For now we are using the pre-install solution and are happy with it.

Many thanks and Regards,
Enrico

ejoerns, 2026-04-13:
I'd consider this resolved (for the `system.conf`) by #1375.

closed 2026-04-13 by ejoerns