rauc/rauc #459: Corrupt system after update

embeddedrocks, 2019-07-02:
Hi there,

Our update bundles consists of two compressed tar files for separate slots. A rootfs and application slot. So when installing the new bundle it will create a new filesystem (format) and then mount the filesystem an untar the slot file. After finishing an unmount is executed. This is executed twice as long as the `slot.raucs` file is different.
According to your documentation and source code that's the defined procedure.

Question: Is there any hidden integrity check that the slot has been written (untar-ed) correctly?

Problem: We had one unit that was successfully updated (all slots), but wasn't able to boot due to corrupt libraries. These libs had been written during the untar process.  So our explanation is that there must have been a write error during the untar operation.

We deployed the update file on many other units without a problem. We also deployed the update file on that unit again and it works perfectly.

Thanks and Best Regards,
Enrico

ejoerns, 2019-07-02:
> Question: Is there any hidden integrity check that the slot has been written (untar-ed) correctly?

No there is no such integrity check. We check that tar returns successfully but that's all.
For a tar unpack it is even not that easy to verify the data was written correctly, as you would have to compare hashes over each individual file.
For a normal image writing this could be done if required, but would also require to make sure actually comparing the data written and not the data cached.
Normally we rely on the underlying layers to properly copy the data.

> Problem: We had one unit that was successfully updated (all slots), but wasn't able to boot due to corrupt libraries.

Would be interesting to know where and when this actually got corrupted. Anyway, this is a good argument for having fallback in the system. ;)

Regards, Enrico

embeddedrocks, 2019-07-02:
> No there is no such integrity check. Normally we rely on the underlying layers to properly copy the data.
That's what we expected and considering the amount of work validating the untar operation fully acceptable.

> Would be interesting to know where and when this actually got corrupted.
Unfortunately there is no further log available to point to the issue.

> Anyway, this is a good argument for having fallback in the system.
Indeed, and the fallback worked perfectly!

But there is a potential problem we actually run into:
As the `rootfs` partition was apparently successfully written during the update, but is actually corrupt, the bootloader switched back to the previous installation. Trying to re-install the same or other bundles (but with the same `rootfs`) won't work as the `rootfs` is never overwritten due to the `slot.raucs` file. Only a new update bundle with a different `rootfs` is going to work as it truly overwrites the slot.
Other option is to create a pre-install hook and delete the `slot.raucs` file.

Many thanks for your prompt reply and Best Regards,
Enrico

ejoerns, 2019-07-02:
> As the rootfs partition was apparently successfully written during the update, but is actually corrupt, the bootloader switched back to the previous installation. Trying to re-install the same or [...]

With https://github.com/rauc/rauc/pull/448 merged, RAUC will disable this behavior by default.
It turned out that this optimization causes too many unexpected behavior. It can still be enabled, but only explicitly via `install-same=false`.

However, also in current versions we have the `force-install-same=true` key that results in the future default behavior and will always override the slot, no matter if the hashes equal or not. But this is only applicable in your case with a two-step update of course.

Using a pre-install slot hook might not work as it won't get executed upon slot skipping.

embeddedrocks, 2019-07-02:
@ejoerns Sounds great, that should fix the issue in general! Probably a good idea to think about updating RAUC to a recent version as we still use quite an old one.

> Using a pre-install slot hook might not work as it won't get executed upon slot skipping.

Good catch - you're right, a pre-install slot hook won't work in our case.

Again, many thanks for your comments and support! Excellent!
Enrico

ejoerns, 2019-07-02:
> @ejoerns Sounds great, that should fix the issue in general! Probably a good idea to think about updating RAUC to a recent version as we still use quite an old one.

It will be in RAUC v1.2, thus not released yet. But updating is always a good idea ;)

> Again, many thanks for your comments and support! Excellent!

Thank you for valuing it! :)

Regards, Enrico

ejoerns, 2019-11-07:
RAUC 1.2 is out now. Thus I assume we can close this ;)

closed 2019-11-07 by ejoerns