rauc/rauc #460: No way to resign when original certificate already expired

mhei, 2019-07-05:
I have a bundle which was signed with a certificate which is meanwhile expired.
I tried to resign the bundle with a new certificate using `rauc resign ...`,
however this fails with
```$ resign-update foobar.raucb
rauc-Message: 12:59:45.721: Reading bundle: foobar.raucb
rauc-Message: 12:59:45.725: Verifying bundle... 
signature verification failed: Verify error:certificate has expired
```

In my opinion, this should always be possible, at least we should provide the ability with a command line parameter. For example it would make sense to me to re-use `--no-verify` here, too.

jluebbe, 2019-07-05:
Yes, we should allow resigning of expired bundles. I'd prefer to have an option like `--allow-exired` to accept just this specific problem, instead of allowing just anything. In addition, `--no-verify` should also work for `rauc resign`.

mhei, 2019-07-06:
I had a quick look and prepared two ideas here: https://github.com/mhei/rauc/tree/issue-460-extend-noverify-resign Please note, that the second commit is totally untested. Also a proper PR should include documentation updates.

Bastian-Krause, 2019-07-10:
@mhei: Your approach looks valid to me. One could probably argue that using the time of signing for `--allow-expired` might not work in all cases (wrong time set on the build host when the bundle is created). To cover this we could set the time for verification to the start or end of the validity period, but I see that your implementation is simpler.

Would you create a PR with your changes?

mhei, 2019-07-10:
My "implementation" is simply (mis-)using existing infrastructure :smile:
I also checked openssl API but I did not find another approach to ignore this specific error but doing full verification on all other aspects. Maybe this could be a feature request for openssl.
I'm a little bit concerned that `--allow-expired` does not exactly reflect what technically is going on when using this approach but it names the use-case so this would be fine with me.
I'll create a PR from my branch as is because next three weeks I will have no time to work on this further, e.g. documentation, sorry.

ejoerns, 2019-07-10:
@mhei Yes.. it does not fully keep the promises it makes.. ;)

And.. OpenSSL is a great API!... :laughing: 

I'm not quite sure if removing a parameter works some how. Setting it to a time that will be valid, as @Bastian-Krause suggested, might work by extracting signature validity from signer cert and using these for validity time, but that sounds hacky. OpenSSL upstream will for sure have a good reason to not allow these things, but maybe one could give it a try.

If I should find some time I'll also have another look at this.

jluebbe, 2019-10-25:
This will be fixed by #498.

jluebbe, 2019-10-27:
Fixed by PR #498.

closed 2019-10-27 by jluebbe