rauc/rauc #463: Failure installing large filesystems on 32bit machines

fboor, 2019-07-17:
A colleague just ran into a problem installing ext4 images > 2GB on a 32bit machine. I took a look at the code and found this: 
The reason seems to be the use of g_output_stream_splice() in update_handler.c, line 134. The documentation of that function indicates that it supports writing files larger than G_MAXSSIZE (which from what I know is limited by the actual size of size_t) but the returned value does not correspond to the actual size processed any more. As a result the size check fails even if the image was written completely.

jluebbe, 2019-07-17:
Yes, checking the return value from g_output_stream_splice() (except for errors) is problematic. We could simply drop the check, but I'd prefer to keep some sort of sanity check there.
I think we should use g_seekable_tell() on the instream as GFileInputStream implements GSeekable.

ejoerns, 2019-11-04:
I've created #503 to address the issue described. Please have a look.

PR #503 (2020-01-09): src/update_handler: use g_seekable_tell() to get size of written data
Documentation of g_output_stream_splice() says:

> Note that if the number of bytes spliced is greater than G_MAXSSIZE,
> then that will be returned, and there is no way to determine the actual
> number of bytes spliced.

As copy_raw_image() performs a sanity check if returned size matches
expected write size, it will not be possible to install an image whose
size exceeds G_MAXSSIZE.

Thus instead of letting g_output_stream_splice() close the output stream, we
leave it open to use g_seekable_tell() on it and obtain the actua [... cut]
commit fbc9e62c (2020-01-09): src/update_handler: use g_seekable_tell() to get size of written data

Documentation of g_output_stream_splice() says:

| Note that if the number of bytes spliced is greater than G_MAXSSIZE,
| then that will be returned, and there is no way to determine the actual
| number of bytes spliced.

As copy_raw_image() performs a sanity check if returned size matches
expected write size, it will not be possible to install an image whose
size exceeds G_MAXSSIZE.

Thus instead of letting g_output_stream_splice() close the output stream, we
leave it open to use g_seekable_tell() on it and obtain the actu [... cut]
closed 2020-01-09 by jluebbe