rauc/rauc #468: License question regarding D-Bus API description file

mhei, 2019-08-01:
The bundled example CGI program in contrib directory shows howto use the D-Bus API
from C programs. It also uses the file `rauc-installer.xml` which is used to
generate code which is then compiled into the resulting binary.

I would like to do the same in a custom (closed-source) project, that means I would like
to include the XML file in my project and use the generated code then. While
I guess that using the generated code is not a problem,
I'm unsure about the API description file itself and its LGPL relation.

What is your opinion?
Is this a real issue?
Should allowed usage be mentioned explicitly in documentation or with a comment in the XML file?
If so, would it be OK to change license of this single file?

jluebbe, 2019-08-01:
Changing the license would be fine with me, but I'd need to check the details. Do you have any examples how other projects are handling this explicitly?

mhei, 2019-08-01:
Unfortunately, no - first case where I would like to use such a API XML file and I also have no other project in mind where similar case could apply...
Another option would be to create a shared library and link custom projects against... boils down to the usual/common LGPL scenario.

jluebbe, 2019-08-01:
As this is the same information you can get from the D-Bus introspection API, I'd tend to view it as simple data.

ejoerns, 2019-11-01:
I assume this clarified the situation and we can close this? Or do we need to document that it can be considered as simple data?

mhei, 2019-11-04:
Yes, the situation is clarified now. I would prefer to have it documented, e.g. by using a XML comment in the first line, see my proposal. Ideally, we could use a SPDX tag in the file, but there seems to be no existing, fitting SPDX id for this.

mhei, 2020-05-28:
I'm closing this issue since we discussed and clarified this topic. Thanks!

closed 2020-05-28 by mhei