rauc/rauc #469: Set up U-Boot Environment for RAUC

mmar2228, 2019-08-07:
Dear all,

I am newbie in RAUC and I have a simple question. I am trying to integrate the boot script selection (contrib/uboot.sh) in u-boot. However, I have problems with tftp and I can not use the command "source" to execute the *.img in u-boot, according to https://www.denx.de/wiki/DULG/UBootScripts.

Is executing all sentences from uboot.sh, step by step, in u-boot prompt a valid option? Is it an alternative?

Is there any relationship between tftp address and /etc/fw_env.config? How I decide the address? I mean, the example command "tftp 100000 /tftpboot/TQM860L/setenv.img" uses 100000 for any specific reason?

Thanks in advance

jluebbe, 2019-08-07:
The script in contrib is only an example. Your version of it should be compiled into the bootloader, so it's not dependent on any (possibly corrupted) storage.
Load commands and addresses are very board-specific in u-boot, so you need to find the correct ones for your board. This is nothing rauc-specific, though.

mmar2228, 2019-08-09:
Thanks very much for your quickly reply! I am sorry for my ignorance.

I will change my uboot coherently, but following the example script. Just a question, I understand from example uboot.sh script that rauc.slot is set to B if BOOT_ORDER="A B" and rauc.slot is set to A if BOOT_ORDER="B A". Therefore, second slot in BOOT_ORDER will be the next active slot. Is it correct?

I assume RAUC defines properly BOOT_ORDER when I do "rauc install ..." in my target. Right?

Thanks again!

ejoerns, 2020-03-04:
@mmar2228 Sorry for the delayed response.

To clarify: No, the script sets `rauc.slot` to the *first* value in `BOOT_ORDER` that has a positive attempts counter.

Note https://github.com/rauc/rauc/blob/master/contrib/uboot.sh#L7 

RAUC sets `BOOT_ORDER` to the right order after installing (with the updated slot being the first element), yes.

mmar2228, 2020-03-13:
Thanks you for your reply.

I explain my doubt just for helping to other people if they make the same mistake than me:

My idea was:

> The  **_for BOOT_SLOT in "${BOOT_ORDER}"; do_** loop does not break if a valid slot is found with first item in BOOT_ORDER. So if BOOT_ORDER="A B", this loop will set rauc.slot=A in first iteration (BOOT_SLOT=A) and then, it will continue with second iteration (BOOT_SLOT=B), and therefore rauc.slot=B is set. 

However, I think that I missed the **_if test "x${bootargs}" != "x"; then_** sentence, which breaks the remaining loop iterations if bootargs has been set. Is it right?

Thanks very much again,

M Mar

ejoerns, 2020-03-13:
@mmar2228 Yes, that's the line I referenced and yes, it aborts the loop as you describe.

Thanks for your feedback, thus I think we can close this issue now.

Regards, Enrico

closed 2020-03-13 by ejoerns