rauc/rauc #470: Question regarding tar.xz image installation without mounting/formatting

odessa2, 2019-08-07:
Hello,
I have a question regarding the capabilities of rauc, or how to achieve a specific task.
The device contains multiple partitions. The partition "data" is mounted in "/partitions/data".

In the rauc config, it is specified as follows:

> [slot.data.0]
> device=/partitions/data
> type=ext4

In the raucm manifest, the image is defined by its filename:
> [image.rootfs]
> filename=rootfs_orig.img
> [image.data]
> filename=data.tar.xz

The rootfs.img is installed to a device with type=raw, this works fine.

When installing the bundle, rauc wants to format the partition using mkfs.ext4, which is not available on the target. How do I tell rauc to just extract the tar.xz in the target directory, and overwrite existing files?

Is it possible to completely take control of the update process of only the data image? Or am I missing an option here? 

Thanks in advance!

jluebbe, 2019-08-07:
> When installing the bundle, rauc wants to format the partition using mkfs.ext4, which is not available on the target. How do I tell rauc to just extract the tar.xz in the target directory, and overwrite existing files?

That is not supported by rauc. The intention is to ensure a fully reproducible state after the update, which is not possible if some files are left over from before the update. So rauc formats the file system to get a defined result.

> Is it possible to completely take control of the update process of only the data image? Or am I missing an option here?

If necessary, you can use slot hooks https://rauc.readthedocs.io/en/latest/using.html#slot-hooks to customize the installation to a slot. I'd be very careful in this case, as this means that the result now depends on which version was installed before the update and each combination would need to be tested.

ejoerns, 2019-10-06:
@odessa2 I assume this answered your question? Can we close this or is there anything left unclear?

ejoerns, 2019-11-01:
As there was no answer and it does not appear to be an issue or something required to support, I'll close this now. Feel free to reopen if this is still relevant for you.

closed 2019-11-01 by ejoerns