rauc/rauc #472: casync support for directory trees

fhackenberger, 2019-08-08:
I can't get rauc with casync to work. I'm using rauc from GIT master@5be3ac48 and tried casync@a8f6c841 and master@{2018-07-05 00:00:00}. Here's my procedure: 

```
$ find bundle-build
bundle-build/
bundle-build/image
bundle-build/manifest.raucm
$ cd bundle-build/
$ cat manifest.raucm
[update]
compatible=rauc-demo-x86
version=2019.08-2
 
[image.rootfs]
filename=board-system-image.catar
$ sudo qemu-nbd --connect=/dev/nbd0 /var/lib/libvirt/images/rauc-test1-slotb.qcow
$ sudo mount /dev/nbd0p1 ~/tmp/board-system-image/bundle-build/image/
$ cd image
$ sudo casync make --verbose ../board-system-image.catar .
$ cd ../..
$ sudo umount bundle-build/image
$ sudo chown ubuntu bundle-build/board-system-image.catar
$ rauc bundle --cert=/home/ubuntu/rauc/demo.cert.pem --key=/home/ubuntu/rauc/demo.key.pem bundle-build system-update.raucb
$ rauc convert --cert=/home/ubuntu/rauc/demo.cert.pem --key=/home/ubuntu/rauc/demo.key.pem --keyring=/home/ubuntu/rauc/demo.cert.pem system-update.raucb system-update.casync.raucb
rauc-Message: 21:16:57.487: Reading bundle: system-update.raucb
rauc-Message: 21:16:57.489: Verifying bundle...
rauc-Message: 21:17:08.210: Converting board-system-image.catar to directory tree idx board-system-image.catar.caidx
tar: This does not look like a tar archive
tar: Skipping to next header
tar: Exiting with failure status due to previous errors
Failed to create bundle: Failed to run casync: Child process exited with code 2
```

I read the rauc source code and I don't get why it tries to extract the .catar file with regular tar: https://github.com/rauc/rauc/commit/80fd1bfec7f3ca66a758372ba5300f894cdd5cd8

Can someone shed some light how to get rauc working with casync on a directory tree?

ejoerns, 2019-08-09:
@fhackenberger the two-step approach for creating a casync bundle is described as

> 1. Create 'conventional' RAUC bundle
> 2. Convert to casync-based RAUC bundle

in the documentation. This means that you do not have to create any kind of casync archives, index files or whatever manually but just create a normal bundle containing either a file system image or an archive (tar).

The approach you did is not wrong on the other hand, but, two problems exist with it:

a) casync does (or did?) not support converting catar to caidx (see https://github.com/systemd/casync/issues/96)
b) handling in casync code is some kind of... wrong

You run into the check function [image_is_archive()](https://github.com/rauc/rauc/blob/master/src/bundle.c#L489) that matches both `*.catar` and `*.tar`.
As a consequence, we call [casync_make_arch()](https://github.com/rauc/rauc/blob/master/src/bundle.c#L111) which takes a tar archive, unpacks it, and runs casync on it. And yes, no matter if you have provided a catar or nor.

Thus, two steps to fix this:

1) Use *.tar image instead of *.catar
2) Remove (comment out) *.catar support as long as not supported #patchopportunity ;)

The golden way would be to ensure casync can handle catar, and support this the in the RAUC code, too.

Regards, Enrico

fhackenberger, 2019-08-09:
Thank you @ejoerns! I think I got how the workflow is supposed to go. I'll give it another shot soon, maybe after my holidays. A patch to improve the ux is also very likely ;-)

ejoerns, 2019-11-01:
@fhackenberger Did you already have time to give it another try? ;)

fhackenberger, 2019-11-12:
@ejoerns Yes, it's working now. One issue I still have is that with my A/B set-up, every update requires (almost) as much storage as there is data on the filesystem, in order to build the chunk store from the seed slot. Is there a way around this?

ejoerns, 2020-03-04:
@fhackenberger The seed slot should not build a chunk store. It should only be used to construct the seed store. This differs from the slot store as it does not store the actual data but only references to where the data can be obtained from.

What instead, is a known 'issue' is that there is no garbage collection or least recently used caching for chunks fetched from remote. They are all stored on the target before being written and kept there.

There are basic plans here on adding handling for this in casync as well as general considerations on also providing a RAUC built-in approach for delta updates that would have this feature.