rauc/rauc #480: rauc 1.1: Pipe file descriptor leak

chrthi, 2019-09-13:
In a system where RAUC 1.1 is started at boot with `/usr/bin/rauc --mount=/run/rauc service`, I observe that a later `rauc install` fails with
```
rauc-Message: 01:23:17.939: installing /tmp/core-bundle-production-foo.raucb: 100% Updating slots failed.
rauc-Message: 01:23:19.022: installing /tmp/core-bundle-production-foo.raucb: 100% Installing failed.
rauc-Message: 01:23:19.059: installing /tmp/core-bundle-production-foo.raucb: LastError: Installation error: Failed marking slot rootfs.1 bootable: failed to activate slot rootfs.1: Failed marking 'rootfs.1' as primary: Failed to start barebox-state: Too many open files
Installing `/tmp/core-bundle-production-foo.raucb` failed
```
When looking through `ls /proc/*/fd/` to investigate the "too many open files", I find that the RAUC service process has lots of pipes opened:
```
lr-x------    1 root     root            64 Aug 25 02:11 7 -> pipe:[7175]
...
lr-x------    1 root     root            64 Aug 25 02:11 953 -> pipe:[661236]
```
I can't find the same pipe opened in any other process.
If I do a `systemctl restart rauc`, the problem goes away and the next install succeeds, but having rauc sit there idling for a longer time seems to be a problem. What is rauc doing during that time that might be leaking pipe fds? I don't really expect it to do anything except listening for D-Bus calls.
Can I somehow provide more helpful information?

ejoerns, 2019-09-16:
@chrthi Thanks for your report. We are currently investigating a similar case where the repeated call of  `rauc status` leaves a growing number of file descriptors open.

Do you perform any D-Bus calls in between? Or really only the install call?

If we make any progress on debugging I will let you know. At the moment I am not yet sure if we see the same or a different issue here.

chrthi, 2019-09-16:
Really only the install call. Which then monitors the status on D-Bus if I understood correctly.
I'm not sure whether the many pipe fds were already present before the `rauc install` call or whether they might have appeared during the installation - then it could have to do with either the status reporting or the subprocesses used for writing images.

ejoerns, 2019-09-16:
Please check if #484 also fixes the issues you see.

From what I've found it would totally make sense if it broke both yours and my scenario with the same cause.

chrthi, 2019-09-17:
Thanks, I've now integrated that fix on my lab units.
In the meantime, I found that my issue was probably caused by [a rauc-mark-good restart loop](https://github.com/rauc/meta-rauc/issues/86). so the fix to `barebox_state_get` could be just the right one. Or could there be another place where `rauc status mark-good` causes a leaking pipe fd?

ejoerns, 2019-09-17:
@chrthi `rauc status mark-good` calling `barebox_state_get()` is very close to what we have seen in an other project, too, yes.

You can check this by executing

    ls -l /proc/$(pidof rauc)/fd | wc -l

having a constant number after calling 'rauc status mark-good'. Without the patch, the number should increase with each call.

ejoerns, 2019-10-02:
@chrthi did that resolve your issues? Then I would close this here.

chrthi, 2019-10-02:
Yes, sorry for the delay. I haven't seen this problem since then.

closed 2019-10-02 by chrthi