rauc/rauc #481: Acknowledge bit in EXT_CSD_PART_CONFIG is reset when flashing eMMC boot partition

s-michaelsen, 2019-09-13:
When flashing a boot partition of an eMMC device the BOOT_ACK (bit 6) bit in EXT_CSD_PART_CONFIG is reset in function r_emmc_write_bootpart().

I'm not sure if this is really the desired behavior. Maybe it would be good to read current status of this bit and keep it or maybe add an appropriate slot option to set it along with partition number?

Of course it could also be set in a script after bootloader was flashed but since missing of this bit might prevent a proper boot on some systems (e.g. TIs AM57xx processors) it might make the update process more reliable if it gets set along with the partition number.

Bastian-Krause, 2019-09-13:
Hi,

I agree, we should preserve the `BOOT_ACK` bit when writing the `PARTITION_CONFIG` register.

Could you create a Pull Request for this?

Regards,
Bastian

s-michaelsen, 2019-09-13:
done ;)

BR,
Stephan

closed 2019-09-13 by s-michaelsen