rauc/rauc #489: Version info to post-install script

jovere, 2019-10-01:
Is there any way to get the bundle version info to the post-install script?

I'd like to be able to write the bundle version information (essentially RAUC_MF_VERSION) to the filesystem, so I know which bundle created it.

Is that possible to do in the present system?  It looks like I'm only getting variables specific to the SLOT in a post-install hook.

ejoerns, 2019-10-02:
@jovere do we talk about hooks or handlers?

For hooks, this information is available, see https://rauc.readthedocs.io/en/latest/using.html#bundle-based-customization-hooks. Is that what you refer to? From the basic design, RAUC is a bit 'per-slottish': But if you enforce update of al slots, the manifest version should be the same for all.
We are also working on more per-installation-based information handling, but that is not implemented, yet.

From what you describe I would assume this is already handled by rauc's slot status file(s). If they are enabled, you should be able to get your info with

    rauc status --detailed

For info about the file see https://rauc.readthedocs.io/en/latest/reference.html#slot-status
Note that this info is also saved on a per-slot base internally.

jovere, 2019-10-02:
Hi Enrico,

I was specifically talking about the slot-post-install hook.  However, it looks like adding a `statusfile` to the system.conf file would fix my issue.  I did not see that in the documentation.

Unfortunately, I'm running the 0.1.1 release, installing through yocto krogoth.  How difficult is it to update to the newest stable release of rauc?

ejoerns, 2019-10-02:
*ringring* 2019 is calling... ;) I really really hope you are not developing for a new system... 

Anyway, RAUC should be fully backwards compatible, thus you should be able to safely bumb version numbers. There might only be changes in packaging, but these should be quite limited.

However, the right way to go is to use recent Yocto version. krogoth is really old...

ejoerns, 2019-11-01:
@jovere did you succeed in updating RAUC / yocto to a recent version?

If your issue is resolved I would like to close this issue.

Thanks and best regards, Enrico

jovere, 2019-11-04:
@ejoerns I was not able to get it updated to the most recent version.  However, I have found a way to get most of what I needed by another means.  This can be closed.

closed 2019-11-04 by jovere