rauc/rauc #490: Possible RAUC bootchooser logic fault

Esco441-91, 2019-10-04:
As the title suggests, I believe there might be a logic fault in the RAUC bootchooser. I'm currently studying how RAUC works and I believe there's a point which has been overlooked in the bootchooser logic in the documentation and the examples. This includes atleast the [Symmetric A/B Setup](https://rauc.readthedocs.io/en/latest/examples.html#symmetric-a-b-setup). Feel free to correct me.

The issue is that the bootchooser logic is always run, even if the system is not being updated. This means that even a normal boot causes the BOOT_<slot>_LEFT to be decremented. This is fine normally, but quick powercycling causes a possible problem. 

As an example raspberry pi, which has around 12 seconds bootup time to the point in which RAUC software marks the slot as "good". If power is cut inbetween bootloader and and this marking, the BOOT_<slot>_LEFT is left decremented. Even if the slot has been running for months. If this happens two more times, the slot will change, assuming that the other slot has been marked as good at some point. This means, that whatever is run may very well be an old, depracated version. Now this situation is something that will happen VERY rarely but I think that there will be those who will come across it, assuming that RAUC will make it into their embedded device distribution. Powercycling by powercompanies are not uncommon, atleast where I come from, and I believe that any technician doing an installation/maintenance work might do this powercycling on site.

If I'm not mistaken, the bootchooser is meant to only detect a faulty partition upon installation of a new image for a slot? Not to analyze the bootstatus of current slot whenever the device is running? I guess this is up for debate, for if the new slot gets corrupted, it's better to have the old slot than no slot.

I'm not an expert on the this software, but I would suggest one of these improvements:

1. Add a flag to bootloader on whenever to run the bootchooser. This way the bootchooser script can be prepended with something like `if ${SLOT_UPDATED}; do`

2. Mark other slots as "bad" upon a succesfull boot. This could also be just a simple instruction in the documentation to guide users to do this 

3. Increase the BOOT_<slot>_LEFT value on a succesfull boot to a high value, something like 1000. This way the bootchooser logic remains but statistically becomes an impossibility on happening by accident. The value should remain as 3 whenever an update is done.

4. Upon first succesfull boot, clone the newly updated slot device to the old one.

Correct me and close the issue if I'm wrong.

ejoerns, 2019-10-05:
Hi @Esco441-91, thanks for your consideration. Maybe we can clarify some things and discuss about the rest.

> If I'm not mistaken, the bootchooser is meant to only detect a faulty partition upon the installation of a new image for a slot? Not to analyze the bootstatus of the current slot whenever the device is running? I guess this is up for debate, for if the new slot gets corrupted, it's better to have the old slot than no slot.

Bootchooser is to guarantee that the selected slot (active slot with the highest prio) can be booted. That there can be multiple attempts to boot the same slot, also in case of failure. And, if booting it really does not work, that another slot is found that one can boot instead.
From the perspective of bootchooser there is really no difference between a slot that was working for ages and a slot that was just updated.

> 1. Add a flag to bootloader on whenever to run the bootchooser. This way the bootchooser script can be prepended with something like if ${SLOT_UPDATED}; do

This is a misunderstanding of bootchooser. There is no script. This is the base idea behind having a built-in framework for this. Bootchooser is in the barebox code and all that needs to be done is the fine-grained configuration of its actual behavior. All the error-prone scripting which is required in U-Boot, GRUB, etc. is avoided.
However, what you propose could also be done from within the code, but it is not the idea of bootchooser to run conditionally. 

> 2. Mark other slots as "bad" upon a successful boot. This could also be just a simple instruction in the documentation to guide users to do this

Sure, this could be done and might also be a requirement in scenarios where fallbacks are not allowed because of security concerns, etc. I would not reject enhancements in the RAUC user documentation concerning these points, but it is also true that configuration of the actual redundancy handling is really use-case specific.

> 3. Increase the BOOT__LEFT value on a successful boot to a high value, something like 1000. This way the bootchooser logic remains but statistically becomes an impossibility on happening by accident. The value should remain as 3 whenever an update is done.

Consider the case that you really run into an issue while booting your slot, even after it was working fine for years. This might be the case because of issues in the underlying file system / storage / etc.
Then you will end up in hours of reboots before the device attempts another slot. This might be an inacceptable downtime!
Having issues in the underlying file system and/or storage is something we must expect when using an image-based update handling. Otherwise, we could also use file-system-tree-based updaters like OSTree.

> 4. Upon first successful boot, clone the newly updated slot device to the old one.

In fact, this is what some people do and what I would propose in a scenario where you need to assure running an updated version but want to gain the ability to fall back to another slot.

Apart from this, I totally agree that the initial boot of a system more 'dangerous' than subsequent boots (maybe not that the second, but the third, fourth, ..). Thus one could argue that the initial boot should have less attempts than subsequent one. But this should be kept in range and this can be handled without changing bootchooser. The RAUC code already slightly reflects this by providing different defines for `BAREBOX_STATE_DEFAULT_ATTEMPS` and `BAREBOX_STATE_ATTEMPS_PRIMARY`. Note that this is not yet highly configurable but allows to set different attempts for initial activation and normal 'mark good'.

Did that answer some of your questions? Are there points you do not agree with?

Esco441-91, 2019-10-07:
Hi @ejoerns,

Thank you for addressing this issue.

> Did that answer some of your questions? Are there points you do not agree with?

You answered my questions well and there's nothing really left to add. I'm glad to learn more perspectives of RAUC usage. If I would request something, it would be to maybe add these perspectives to the documentation atleast? 

Regardless of what Your future actions will be, my issue/question is resolved. Mostly I wanted to bring these things mentioned to light as I saw a possible fault. Fell free to close/resolve this issue.

closed 2019-10-07 by Esco441-91