rauc/rauc #491: Process de.pengutronix.rauc received signal 5

sjs205, 2019-10-07:
Hello, 

I'm trying to integrate RAUC into my yocto build for a sunxi based system. So far integration is going well, except I am unable to to mark the rauc status as good with:

```
$ rauc status
rauc-Message: 13:31:51.894: rauc mark: rauc status: error retrieving slot status via D-Bus: error creating proxy: Error calling StartServiceByName for de.pengutronix.rauc: GDBus.Error:org.freedesktop.DBus.Error.Spawn.ChildSignaled: Process de.pengutronix.rauc received signal 5
```

I have followed the documentation regarding integrating into d-bus, and my ```/usr/share/dbus-1/system.d/de.pengutronix.rauc.conf``` file is in place. I can't see the ```de.pengutronix.rauc.service``` file in the ```data``` directory so I assume that the version I'm using does not include this.

I've no idea how to debug the D-Bus service. 

Any idea what is going on here? Or tips to debug?

Notes:
I'm on the ```origin/warrior``` branch
version 1.1

Many thanks

ejoerns, 2019-10-07:
@sjs205 glad you resolved this. However, not having 'service' in help is something I've never noticed and which I don't think we did intentionally,
Thanks for reporting, I've created #492 to fix this

sjs205, 2019-10-07:
@ejoerns , many thanks. Happy to help :)