rauc/rauc #493: "No keyring file or directory provided" when trying to install a new bundle

sjs205, 2019-10-09:
Hello again,

I'm trying to install a bundle on my new board setup, but I'm having some problems with the ```rauc install``` command. 

Lets start with the ```rauc info``` command. If I run this without the ```--keyfile``` argument I get the following error:
```
root@olinuxino-a20:~# rauc info /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb
No keyring file or directory provided
```

Obviously, I can fix this by adding the keyfile:
```
root@olinuxino-a20:~# rauc --keyring=/etc/rauc/ca.cert.pem info /mnt/tmp/core-bu
ndle-minimal-olinuxino-a20.raucb
rauc-Message: 16:30:35.905: Reading bundle: /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb
rauc-Message: 16:30:35.908: Verifying bundle... 
Compatible:     'myDev1.5'
Version:        '1.0'
Description:    'core-bundle-minimal version 1.0-r0'
Build:          '20191008130844'
Hooks:          ''
1 Image:
(1)     core-image-minimal-olinuxino-a20.ext3
        Slotclass: rootfs
        Checksum:  1a799e0c65fbeb7ae3c54526d397b47d5b5c458852bea93b0e961f19a00a8608
        Size:      39845888
        Hooks:     
0 Files

Certificate Chain:
 0 Subject: /O=Test Org/CN=Test Org Development-1
   Issuer: /O=Test Org/CN=Test Org rauc CA Development
   SPKI sha256: CB:49:2C:6D:F3:15:72:2B:2C:F0:F3:E5:D6:A0:70:9B:D1:FD:62:32:9C:F0:76:74:63:36:68:DD:B7:D0:79:B3
   Not Before: Jan  1 00:00:00 1970 GMT
   Not After:  Dec 31 23:59:59 9999 GMT
 1 Subject: /O=Test Org/CN=Test Org rauc CA Development
   Issuer: /O=Test Org/CN=Test Org rauc CA Development
   SPKI sha256: AC:6C:35:B7:23:73:0B:76:C9:EE:D3:77:E7:AD:4D:FC:4C:BE:BB:37:F9:79:7A:86:26:7E:11:28:CB:5F:89:2A
   Not Before: Jan  1 00:00:00 1970 GMT
   Not After:  Dec 31 23:59:59 9999 GMT
```

But this does not appear to be the case with the install command. Without cert:
```
root@olinuxino-a20:~# rauc install /mnt/tmp/core-bundle-minimal-olinuxino-a20.ra
ucb
rauc-Message: 16:47:56.611: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb: installing
rauc-Message: 16:47:56.618: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb:   0% Installing
rauc-Message: 16:47:56.619: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb:   0% Determining slot states
rauc-Message: 16:47:56.622: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb:  20% Determining slot states done.
rauc-Message: 16:47:56.624: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb:  20% Checking bundle
rauc-Message: 16:47:56.628: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb:  40% Checking bundle failed.
rauc-Message: 16:47:56.629: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb: 100% Installing failed.
rauc-Message: 16:47:56.629: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb: LastError: No keyring file or directory provided
rauc-Message: 16:47:56.630: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb: idle
Installing `/mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb` failed
```

With cert:
```
root@olinuxino-a20:~# rauc --keyring=/etc/rauc/ca.cert.pem install /mnt/tmp/core
-bundle-minimal-olinuxino-a20.raucb
rauc-Message: 16:48:27.385: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb: installing
rauc-Message: 16:48:27.391: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb:   0% Installing
rauc-Message: 16:48:27.393: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb:   0% Determining slot states
rauc-Message: 16:48:27.396: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb:  20% Determining slot states done.
rauc-Message: 16:48:27.397: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb:  20% Checking bundle
rauc-Message: 16:48:27.402: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb:  40% Checking bundle failed.
rauc-Message: 16:48:27.402: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb: 100% Installing failed.
rauc-Message: 16:48:27.403: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb: LastError: No keyring file or directory provided
rauc-Message: 16:48:27.403: installing /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb: idle
Installing `/mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb` failed
```

Which leaves me wondering. How do I install the bundle?

ejoerns, 2019-10-09:
@sjs205 This does not work because of the (default) split of RAUC as a service and a client part.

When you pass --keyring to the client, this will not change how the service checks incoming bundles.

At the moment we do not (and cannot (yet)) forward arguments to the install process (see #446).

The right handling would be to set a proper path to the keyring in your system.conf file (`[keyring]` section), see https://rauc.readthedocs.io/en/latest/reference.html#system-configuration-file

ejoerns, 2019-10-09:
https://rauc.readthedocs.io/en/latest/using.html#creating-bundles

Don't you use a build system? This should automate things a little..

sjs205, 2019-10-09:
Yes, I'm using Yocto. I believe there may be a problem with my setup. But I've not seen the manifest file as an output. I'm able to generate the bundle with:

```
bitbake core-bundle-minimal
```

The bit that isn't clear is the ```No such file or directory```. Should I be passing an argument with this manifest file? Or should I copy this to the target?

ejoerns, 2019-10-09:
@sjs205 When does this error appear? 

It should not appear on the target as this can only originate from either 'rauc bundle' or 'rauc checksum' calls. Thus how do you get this on the target? What do you execute to get it?

sjs205, 2019-10-09:
DOH! And there it is. The correct command is ```rauc install```, I was running ```rauc bundle install```

This was my command and the debug output:
```
root@olinuxino-a20:~# rauc bundle install /mnt/tmp/core-bundle-minimal-olinuxino
-a20.raucb  --key=/mnt/tmp/development-1.key.pem --cert=/mnt/tmp/development-1.c
ert.pem  --debug
rauc-Message: 18:30:35.380: Debug log domains: 'rauc'

(rauc:389): rauc-DEBUG: 18:30:35.388: bundle start
(rauc:389): rauc-DEBUG: 18:30:35.389: No value for key "max-bundle-download-size" in [system] defined - using default value of 8388608 bytes.
(rauc:389): rauc-DEBUG: 18:30:35.389: No mount prefix provided, using /mnt/rauc/ as default
(rauc:389): rauc-DEBUG: 18:30:35.390: input directory: /home/root/install
(rauc:389): rauc-DEBUG: 18:30:35.390: output bundle: /mnt/tmp/core-bundle-minimal-olinuxino-a20.raucb
Failed to update manifest: No such file or directory
```

sjs205, 2019-10-09:
Apologies and thank you for your help.

closed 2019-10-09 by sjs205

ejoerns, 2019-10-09:
:laughing: :+1: