rauc/rauc #494: Images with ext2 and ext3 extensions do not appear to be supported

sjs205, 2019-10-09:
Me again! :)

I was trying to update a bundle that used the ext2/ext3 filesystems, and so my Yocto generated filesystem images had the {{{.ext2}}} or {{{.ext3}}} extensions.

When I tried to update the bundles, I was getting errors similar to the following:
```
0.raucb: LastError: Installation error: Unsupported image /mnt/rauc/bundle/core-image-minimal-olinuxino-a20.ext3 for slot type ext3
```

I've checked the source and believe this occurs because the ```RaucUpdatePair updatepairs``` in ```src/update_handler.c``` does not allow for ext2 or 3. Is this intentional?

In the end, I simply generated an ext4 based images to get around this.

ejoerns, 2019-10-09:
@sjs205 I cannot see any reason why one would want to use ext2 or ext3. They are more ore less a subset of ext4, thus we did not see any point in still supporting them nowadays.

If there are no good reasons against ext4, I would say using ext4 is the right way to go.

sjs205, 2019-10-09:
As I understand it, ext2 is used when one specifically does not require (or want) journaling. In my use-case, we are trying to reduce writes to disc, and hence, a non-journaling FS is what we need. 

That said, my understanding may be wrong, or there may be better ways to achieve this.

All things considered, is this ```RaucUpdatePair updatepairs``` just a sanity check array, and hence, to add support for these file systems, we'd simply need to add these cases here? So for a few lines, more FS would be supported?

sjs205, 2019-10-09:
Seems journaling can be disabled quite easily: https://foxutech.com/how-to-disable-enable-journaling/

Maybe it is better to enforce ext4...

ejoerns, 2019-10-09:
> All things considered, is this `RaucUpdatePair updatepairs` just a sanity check array, and hence, to add support for these file systems, we'd simply need to add these cases here? So for a few lines, more FS would be supported?

Yes. It is not a question of feasibility.

> Seems journaling can be disabled quite easily: https://foxutech.com/how-to-disable-enable-journaling/

Yes, this is what I meant by subset. I don't know of anything that ext2/3 can do that ext4 cannot be configured to do.

> Maybe it is better to enforce ext4...

That's what I wanted to point out ;)

sjs205, 2019-10-09:
The only use-cases I can think of is where one would not want to include ext3 or ext4 drivers in the final system, maybe for reducing image size, especially since ext2 is perfectly adequate for many systems. Also, ext4 requires additional configuration. Maybe there are instances where kernel recompilation isn't an option. 

Anyway, I'll close this ticket. :)

closed 2019-10-09 by sjs205