rauc/rauc #496: Is it feasible to implement support for timestamping servers?

thirtythreeforty, 2019-10-14:
A problem with signed RAUC update images is that over time, their underlying certificate will expire, turning them into useless blobs (until a human intervenes and re-signs them), even though they are no less "valid" than when they were made.  In addition, it is not possible to update a system that does not have a valid system time -- again, the certificate is not (yet) valid.

`use-bundle-signing-time` mitigates this somewhat by making the _signing time_ the relevant time that is checked.  Using this makes the above problems go away, in exchange for a new one: it is not really possible to have certificates expire, since someone with a cert that has expired could simply roll back their clock when making a signature.

The usual method to finish mitigating this is to use a timestamping server, who is a trusted authority who cryptographically attests that a certain signature exists at a certain time.

I am not terribly familiar with CMS, but it seems to have slots for more than one signature.  Would validating timestamps in this way be feasible to implement?

jluebbe, 2019-10-15:
Even with a time-stamping service, your systems would need to have some source for a valid system time. There's no way around that.

As far as I understand it, timestamped signatures are possible with CMS. The problem here is that this only moves the issue again: You now need to trust that the time-stamping key will not be compromised or expire "too early".
In my opinion, `use-bundle-signing-time` results in almost the same behavior as a non-expiring signing certificate: If it's key is compromised at any point, the attacker can produce valid bundles.

By using expiring certificates for signing, you limit the point in time where bundles can be *installed*, not *created*. This limits the risk of rollback attacks and also the length of time during which a certificate could be used by an attacker. So with that perspective, the explicit re-signing with a new key means a re-approval that this specific bundle is still allowed as an update (i.e. regarding security vulnerabilities discovered in the meantime). In that sense, they *are* less "valid" than when they were made.

At least for updates via the network, instead of using timestamps in the bundle itself, you could alternatively enforce downloading the bundle via HTTPS, which authenticates the server at the moment of the download.

So I think the only scenario where time-stamping is useful is:
- You trust that the time-stamping third party protects their key significantly better than you can protect your signing keys.
- Your signing keys are rotated relatively quickly or you are confident that you detect compromise quickly and can revoke them (otherwise the attacker might be able to still get a valid timestamp).
- The bundles should be installed only based on whether they were signed correctly, not based on how old they are.

That said, I'm not against implementing time-stamped signing support, but it should come with a convincing use-case. :)
An alternative to consider could be requiring N-of-M (non-expiring) signatures. Multiple signatures are already supported by the CMS format and OpenSSL, but not by RAUC yet..

thirtythreeforty, 2019-10-15:
> In my opinion, `use-bundle-signing-time` results in almost the same behavior as a non-expiring signing certificate: If it's key is compromised at any point, the attacker can produce valid bundles.

This is true!

> The problem here is that [timestamping] only moves the issue again: You now need to trust that the time-stamping key will not be compromised or expire "too early".

Well, since you trust that the timestamping server is an authority about time, I don't see why you couldn't use _its_ signing time as the time to check, and not have to worry about its certificate expiry either.  Then your only concern about the timestamping server is whether the server's certificate is compromised.  These services are often trust anchors, so I don't think that's a terribly large concern.

For our use-case, the end user will be interacting with firmware images, and they will expect them to act like an OS installation CD - they should not expire due to signing concerns.  Moreover, the system in question will never have internet access - I agree that if you do have internet, there are better options than this.

jluebbe, 2019-10-17:
> Well, since you trust that the timestamping server is an authority about time, I don't see why you couldn't use _its_ signing time as the time to check, and not have to worry about its certificate [...]

So you're talking about a third-party time-stamping service, right? Would you accept timestamps from any standard brower CA or just a specific subset? If they discontinue their service, you wouldn't be able to sign updates anymore.

> For our use-case, the end user will be interacting with firmware images, and they will expect them to act like an OS installation CD - they should not expire due to signing concerns, vulnerabilities and all. Moreover, the system in question will never have internet access - I agree that if you do have internet, there are better options than this.

OK. Makes the design space much smaller. :)

So you'd be using an additional timestamp to make bundles signed by short term certificates installable forever.
An alternative might be to have non-expiring certificates, but keep revoking (via CRL in the image) and replacing them regularly (perhaps per major version). Then you'd also have the benefit of rejecting downgrades.

If you think it's worth implementing this and it doesn't make the verification code more complicated, it'd be willing to accept a PR.

thirtythreeforty, 2019-10-17:
> What do you think of requiring multiple valid signatures to reduce the risk of key compromise?

That's an interesting option.  As a side effect you get N-of-N (i.e. all 4 people must sign).

> An alternative might be to have non-expiring certificates, but keep revoking (via CRL in the image)

Yes, but I think our users won't be doing maintenance that frequently, and firmware upgrades do require a small amount of intervention.

> it'd be willing to accept a PR.

OK thanks - I will see what our customer wants and I can definitely work on this if that's the route we're going.