rauc/rauc #497: Update a single file within rootfs

reyhaney84, 2019-10-15:
Hello everyone,

The RAUC documentation describes:

_In RAUC, everything that can be updated is a slot.  Thus a slot can either be a
full device, a partition, a volume or simply a file._

Now, I have two systems: system0 and system1, each of them defined as a
partition on eMMC.

I want to update a single file in my rootfs, for example the binary file of the
application. How should I setup my system.conf and manifest?

I could update the whole partition as an image, but just one file, no success.
I tried to follow this link: https://github.com/rauc/rauc/issues/114#issuecomment-300088080
but I always get the error: Installation error: Unsupported image /run/rauc/bundle/test for slot type file.

what should be the type in manifest and system.conf? Just assume for a simple test, it is a text file that should be updated.

I would appreciate, if someone could help me.

Thanks.

jluebbe, 2019-10-15:
Ah, that is a leftover from the (now-removed) network download mode, which worked without bundles and was not really used.

If you use `type=raw`, RAUC won't care that the the file is not a real device and write to it. Note that this is not intended to be used for updating files in *another slot*, we've used it for large data files on a separate partition.

What is your larger use case for updating a single file? RAUC is built around the conceptual model of a bundle fully describing the intended target state of the system.

If this is something you want to do regularly and want to reduce the download size, you could take a look at casync bundles. Alternatively, we've been thinking about implementing a incremental bundle mechanism, which would create a "delta" from two "full" bundles. But for now, this is not really on our road-map. If you're interested in that, I could explain in detail how that could be implemented.

If this is a one-off emergency fix, you could do a full custom update, which basically just runs a script from the bundle: https://rauc.readthedocs.io/en/latest/using.html#full-custom-update

reyhaney84, 2019-10-15:
This is my manifest:
```
[image.config] 
filename=test
```

And this is my system.conf:
```
#System A
[slot.rootfs.0]
device=/dev/mmcblk1p5
type=ext4
bootname=system0

[slot.config.0]
device=/test
type=raw

#System B
[slot.rootfs.1]
device=/dev/mmcblk1p6
type=ext4
bootname=system1

[slot.config.1]
device=/test
type=raw
```
By a short test, I still receive the same error: 

> LastError: Installation error: Unsupported image /run/rauc/bundle/test for slot type raw

Also, when I check the path /run/rauc/, there is empty. Currently I am on system1, and want to update test file in system0. 

Actually, I am thinking about two different scenarios for my use-case:
**Scenario1:**

- System0: partition1 which contains rootfs,kernel and application
- System1: partition2 which contains rootfs, kernel and application

**Scenario2:**

- System0:

 
partition1: which contains rootfs,kernel
partition2: which contains application

- System1: 

partition3: which contains rootfs,kernel
partition4: which contains application

The point is, application will be updated frequently in comparison to the rootfs. So, if I am going to use the first scenario and update bundle, then it means I should always update a large partition and update time will increase. Because of that, I wanted to test updating single file within rootfs.

By the second scenario, there will be no problem: because I will update application partition as a bundle, and its size is less than rootfs partition.

Which scenario do you recommend?

ejoerns, 2019-11-01:
@reyhaney84 What @jluebbe suggested are the basic options that I would recommend in this case, too.

As he already pointed out, the conceptual background of requiring a bundle to describe the entire system is to always have a full well-defined state of all components in the exact state and assembly as they were tested before deploying.

Thus the 'misuse' of slots to install separate files and gain some kind of delta update is something I would strongly encourage you not to do.

But maybe I can point out some other options or clarify things.

When your actual motivation is to save update time, then I would first of all assume that it is not a question of bundle size but more a question of writing time, right?

In case of Scenario2, we have the optimization in RAUC to skip the installation of images to slots whose current content already equals the one to be installed. This is referred to as 'slot skipping' in the documentation [1].
By relying on this (requires an explicit `install-same=false` in newer RAUC versions) you can gain installation speed for installing your system multiple times with an unchanged rootfs image.

The further optimization of this (when bundle size should actually be an issue, too) would be to not include the content (image) of the rootfs (that is the same anyway) in the bundle and only write the expected rootfs checksum to the slots classes manifest section. This way the installation will abort in case the expected rootfs checksum should differ from the actual one.
Note that this behavior is not (fully) implemented, yet.

Regards, Enrico

[1] https://rauc.readthedocs.io/en/latest/basic.html#slot-status-and-skipping-slot-updates

ejoerns, 2020-03-04:
@reyhaney84 did this clarify things and resolve your issues? Then I'd like to close this issue.

closed 2021-01-08 by ejoerns