rauc/rauc #506: Avoid dbus dependency

jluebbe, 2019-11-06:
Since 7698cadcbbd0db63877a2947b1d6b9cc7b556f58 we have a strict dependency on the dbus-1 pkg-config module. @michaelolbrich suggested to only search for it if `--with-dbussystemservicedir=DIR` is not given on the configure command-line and we actually need to do the auto-configuration. When this is fixed, the workaround in meta-rauc should be removed (see https://github.com/rauc/meta-rauc/pull/92).

jluebbe, 2020-03-10:
@michaelolbrich In https://github.com/jluebbe/rauc/commit/d1e0914e875cc10228c6c58b026934c5ce0c804b, I tried to implement this. Is seems to work (no `checking for DBUS1... yes` when the `-with-*=DIR` options are given), but if they are not, it searches for DBUS1 three times. Is there a better way?

jluebbe, 2022-01-26:
Instead of solving these minor issues with our autotools setup, I'd prefer to close this and move to Meson (#829). @ejoerns, what do you think?

ejoerns, 2022-01-26:
@jluebbe time is probably better invested in meson support as long as this does not cause any real-world pain (which it does not seem to).

I'll just close this as this was idle for quite a long time already, anyway.

closed 2022-01-26 by ejoerns