rauc/rauc #509: Bundle verification error

Ygenks, 2019-11-12:
Hello,

I'm using rauc integrated with yocto using meta-rauc. 
When I'm trying to run `rauc info` on generated image I get the following error:

`$ rauc info tseb-system-update-tseb-am5728-con-20191112085825.raucb`
```rauc-Message: Reading bundle: /tseb-system-update-tseb-am5728-con-20191112085825.raucb
rauc-Message: Verifying bundle...
cryptodev_digest_final: digest failed
3069830880:error:2E09D06D:CMS routines:CMS_verify:content verify error:cms_smime.c:393:
signature verification failed: error:2E09A093:CMS routines:CMS_SignerInfo_verify_content:unable to finalize context
```

Maybe you have any ideas why this happening? 

The package versions I'm using: 
rauc 1.1 
meta-rauc on morty b61c96b881c27ede9c7fc10a52a3782d79d5e2a9
OpenSSL 1.0.2j  26 Sep 2016

Here is my setup:

system.conf:
```
[system]
compatible=tseb-am5728
bootloader=uboot

[keyring]
path=/etc/rauc/ca.cert.pem

[slot.rootfs.0]
device=/dev/mmcblk0p2
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/mmcblk0p3
type=ext4
bootname=B
```

Bundle generation recipe:
```
inherit bundle

RAUC_BUNDLE_COMPATIBLE ?= "tseb-am5728-con"

RAUC_BUNDLE_SLOTS ?= "rootfs"

RAUC_SLOT_rootfs ?= "tseb-image-dev"
RAUC_SLOT_rootfs[fstype] ?= "ext4"

RAUC_KEY_FILE = "${SOURCES_DIR}/meta-tall/files/key.pem"
RAUC_CERT_FILE = "${SOURCES_DIR}/meta-tall/files/cert.pem"
```
The way I generate certificates:

1. Run  [openssl-ca.sh](https://github.com/rauc/meta-rauc/blob/master/scripts/openssl-ca.sh)  provided in meta-rauc layer
2. Put ca.cert.pem from openssl-ca/dev to /etc/rauc/
3. Sign rauc bundle with openssl-ca/dev/development-1.cert.pem and openssl-ca/dev/private/development-1.key.pem which are placed according to .bb recipe

jluebbe, 2019-11-12:
> I'm using rauc integrated with yocto using meta-rauc.
> When I'm trying to run `rauc info` on generated image I get the following error:
> 
> `$ rauc info tseb-system-update-tseb-am5728-con-20191112085825.raucb`
> 
> ```
> rauc-Message: Verifying bundle...
> cryptodev_digest_final: digest failed
> 3069830880:error:2E09D06D:CMS routines:CMS_verify:content verify error:cms_smime.c:393:
> signature verification failed: error:2E09A093:CMS routines:CMS_SignerInfo_verify_content:unable to finalize context
> ```
> 
> Maybe you have any ideas why this happening?

It seems that the hash calculation itself fails in OpenSSL (`cryptodev_digest_final: digest failed`). Mainline OpenSSL doesn't support cryptodev, so it seems you're using a patched version to get hardware acceleration? Please try without cyptodev, I've seen many interesting problems with cryptodev. ;)

> The package versions I'm using:
> rauc 1.1
> meta-rauc on morty b61c96b881c27ede9c7fc10a52a3782d79d5e2a9
> OpenSSL 1.0.2j 26 Sep 2016

Note that both morty and OpenSSL 1.0.2j are old and superseded. The OpenSSL 1.0.2 branch will be EOL at the end of this year!.

Ygenks, 2019-11-15:
Many thanks!
Really the problem was in cryptodev modules for OpenSSL.

jluebbe, 2019-11-15:
Depending on where you got your cryptodev implementation (silicon vendor?), you might want to report that bug to them.

closed 2019-11-15 by jluebbe