rauc/rauc #511: casync chunks the seed chunk store

fhackenberger, 2019-11-14:
RAUC with cassync seems to create a chunk index of the chunk index when installing a RAUC bundle.

```
root     29897  0.0  0.1 256164 10172 pts/0    Sl   18:36   0:00 rauc -d install /tmp/board-system-images-v3/update.casync.raucb
root     29922 85.9  0.1  48132 14444 ?        R    18:36  11:15 casync --log-level=debug --verbose extract --seed /run/rauc/rootfs.0 --store /tmp/board-system-images-v3/update.casync.castr --seed-output=no /run/rauc/bundle/rootfs.tar.gz.caidx /run/rauc/rootfs.1
```

```
ubuntu@sign-pc-18:~$ sudo du -xhcs /var/tmp/*.cased | sort -n     
3.3G    total
3.3G    /var/tmp/3a875a5021649923.cased
Logs from casync:
```

```
Nov 12 18:48:54 sign-pc-18 rauc[29863]: Seeding var/tmp/3a875a5021649923.cased/ec82/ec828c3b0e9ca9277f2b36051edce47e8cae0e128e3fb21aeb70de8978c0ae99
Nov 12 18:48:54 sign-pc-18 rauc[29863]: Seeded var/tmp/3a875a5021649923.cased/ec82/ec828c3b0e9ca9277f2b36051edce47e8cae0e128e3fb21aeb70de8978c0ae99
```

fhackenberger, 2019-11-19:
The solution for the bug is quite easy:
Write a .caexclude file with the content:
`/var/tmp/*.cased`
and store it in `/` before running casync. This should happen in `casync_extract` in src/update_handler.c if a `seed` is set. Maybe there is also a way to make that path explicit when calling casync, from what I understand casync creates a new random directory for the extract run, if `--seed` is specified.

fhackenberger, 2019-11-19:
See https://github.com/rauc/rauc/pull/513 for a workaround, where a user created .caexclude file can be used to fix the initial `casync make` run.

ejoerns, 2020-01-03:
Maybe the right option is to set TMPDIR to a device not being the seeded one? Should be combined with `--exclude-submounts=yes` if I get it right.

fhackenberger, 2020-01-06:
In our current use case, with a large base image, having a separate tmpdir would create the need for a third large partition and would leave the issue with the very slow installation unresolved. So we went for the use-case I described in the PR. I'll close this now thanks for your feedback!

closed 2020-01-06 by fhackenberger

ejoerns, 2020-01-06:
Maybe it is worth documenting the different RAUC/casync use cases together with their needs, pros and cons.
(And also write some hints on TMPDIR storage location).