rauc/rauc #517: confusing error message when dowanloading full bundle over HTTP

jktjkt, 2019-12-04:
This is really a duplicate of #352, but I did not find that one before. When I'm downloading a bundle that is larger than the default (8MB), the error message points to a file write error instead of identifying that it's really about a file being larger than an internal default value:

```
LastError: Failed to download bundle http://example.org/tmp/update.raucb: Transfer failed: Failed writing body (0 != 1448)
```

I would like the error message to be specific, i.e. to point out to that config option.

ejoerns, 2020-01-06:
I've created #522 to address this issue. Would that be sufficient in your opinion?

jktjkt, 2020-01-06:
Thanks. It's definitely an improvement. It does not point to the docs on how to change this (which is just one `if` away), but it will make it obvious that the problem is related to some file size limit -- and then googling for "rauc file size limit" shows a relevant page from the docs as the first hit for me.

LGTM.

ejoerns, 2020-03-04:
I'd consider this basically fixed and close this issue. @jktjkt if you have a good suggestion on how to improve the error message, feel free to open another PR for this.

Thanks for the general hint!

closed 2020-03-04 by ejoerns