rauc/rauc #520: Bundle contains files not in manifest

Psykar, 2019-12-10:
If I run `rauc bundle` and point it at a directory containing multiple versions of the filesystem (just in different formats) the resulting bundle ends up including ALL of them, rather than just the file mentioned in the manifest.

eg: a ext4 file of 60MB ends up making a bundle of 280MB!
```
[update]
compatible=someboard
version=2019.12.10-1

[image.rootfs]
sha256=246f83255264c564c9dd3e0d6017841aa4b7f3ecab14a42daa67082390d5ae66
size=67742720
filename=rootfs.ext4
```

```
$ ls images/linux/ 
BOOT.BIN         rootfs.cpio.gz.u-boot  rootfs.jffs2          system.bit        vmlinux
image.ub         rootfs.ext3            rootfs.manifest       system.dtb        zImage
manifest.raucm   rootfs.ext3.bz2        rootfs.tar.bz2        System.map.linux  zynq_fsbl.elf
rootfs.cpio      rootfs.ext4            rootfs.tar.gz         u-boot.bin
rootfs.cpio.bz2  rootfs.ext4.gz         rootfs.testdata.json  u-boot.elf
rootfs.cpio.gz   rootfs.ext4.zst        sdcard/               uImage
$ ls -l images/linux/rootfs.ext4
-rw-r--r-- 1 <censored> 1000 67742720 Dec 10 13:42 images/linux/rootfs.ext4

$ rauc bundle --cert=project-spec/meta-rauc/scripts/openssl-ca/dev/development-1.cert.pem   --key=project-spec/meta-rauc/scripts/openssl-ca/dev/private/development-1.key.pem  ./images/linux/ bundle.raucb`

ls -l bundle.raucb
rw-r--r-- 1 <censored> users 294344214 Dec 10 16:00 bundle.raucb
```

(FWIW I realize this looks like I'm using yocto and should just use the bundle recpie, but I'm actually using zylinx's petalinux which is a yocto wrapper and [so far] isn't playing nicely with the bundle wrapper...)

jluebbe, 2019-12-10:
It is intentional that rauc bundles all files in the directory. This is useful for example for hook scripts. You need to run `rauc bundle` on a directory containing only the files you want to ship.

jluebbe, 2019-12-13:
Solved via #521, thanks!

closed 2019-12-13 by jluebbe