rauc/rauc #531: Pass running system version to handler/hooks

obbardc, 2020-02-01:
It would be great to pass the running system version to hooks, specifically the `handlers.pre-install` hook and the `install-check` bundle hook.

This would allow the bundle and system hook scripts to do basic checking against the installed version and version to install.

Can you see any issue with passing this information to the hook scripts?
With your blessing, I will do a pull request to implement this feature.

ejoerns, 2020-02-01:
@obbardc to clarify, with *system version* you refer to the rauc bundle 'version' information stored in extended per-slot state?

This would be great to have but probably requires some more changes to ensure always having a valid one available.

I'm currently working on something similar but that is not fully ready yet.

Or did you mean a different system version?

obbardc, 2020-02-01:
Yeah, exactly that. Pass the version of the current running slot (and all
other slots too, maybe) to the hook.

Do you have planned a branch?

Best wishes

ejoerns, 2020-03-04:
@obbardc Yes, I was talking about https://github.com/ejoerns/rauc/commits/topic/slot-status-concept

obbardc, 2020-03-04:
Branch is looking good, however my use-case is a little bit different: making sure that when upgrading to a new bundle that it really is an upgrade (i.e not a downgrade or re-installing the current version).

For that we use an incremental bundle version and the following patch which passes the current system version to the bundle hook in an envvar. The bundle hook script then decides whether the bundle should be installed or not.
This envvar could be passed to the pre-install system script also.

Hopefully my usecase makes sense and may be useful to others?
In its current state my patch is hacky, but I'm interested in getting a solution merged in RAUC.

```
From b4c44da9c355cf8ce7aa49dd14b7a3805f1f9779 Mon Sep 17 00:00:00 2001
From: Christopher Obbard <chris.obbard@collabora.com>
Date: Tue, 4 Feb 2020 13:12:55 +0000
Subject: Pass system version to bundle install hooks

Pass installed system version to bundle install-check hook script.

---
 src/install.c | 13 +++++++++++++
 1 file changed, 13 insertions(+)

diff --git a/src/install.c b/src/install.c
index 3e5e44d..dc52e64 100644
--- a/src/install.c
+++ b/src/install.c
@@ -601,6 +601,8 @@ static gboolean run_bundle_hook(RaucManifest *manifest, gchar* bundledir, const
 	g_autoptr(GDataInputStream) datainstream = NULL;
 	gboolean res = FALSE;
 	gchar *outline, *hookreturnmsg = NULL;
+	g_autofree gchar *system_version = NULL;
+	RaucSlot *s = NULL;
 
 	g_assert_nonnull(manifest->hook_name);
 
@@ -608,10 +610,21 @@ static gboolean run_bundle_hook(RaucManifest *manifest, gchar* bundledir, const
 
 	g_message("Running bundle hook %s", hook_cmd);
 
+	/* pass the version of the current slot to the hooks script */
+	/* a proper method is TBD upstream, see: https://github.com/rauc/rauc/issues/531 */
+	s = find_config_slot_by_device(r_context()->config, r_context()->bootslot);
+	if (s) {
+		load_slot_status(s);
+		if (s->status)
+			system_version = g_strdup(s->status->bundle_version);
+	}
+	g_message("bootslot=%s, system_version=%s", r_context()->bootslot, system_version ?: "");
+
 	launcher = g_subprocess_launcher_new(G_SUBPROCESS_FLAGS_STDERR_PIPE);
 
 	g_subprocess_launcher_setenv(launcher, "RAUC_SYSTEM_COMPATIBLE", r_context()->config->system_compatible, TRUE);
 	g_subprocess_launcher_setenv(launcher, "RAUC_SYSTEM_VARIANT", r_context()->config->system_variant ?: "", TRUE);
+	g_subprocess_launcher_setenv(launcher, "RAUC_SYSTEM_VERSION", system_version ?: "", TRUE);
 
 	g_subprocess_launcher_setenv(launcher, "RAUC_MF_COMPATIBLE", manifest->update_compatible, TRUE);
 	g_subprocess_launcher_setenv(launcher, "RAUC_MF_VERSION", manifest->update_version ?: "", TRUE);
```

ejoerns, 2020-03-04:
@obbardc The issue to be solved first is the same for you use case, too. We need to provide a global information about what is currently running on the system.

The hacky approach here goes into the same direction but won't fit the overall concept. You already see that you put status information which is slot-specific into a global context where only static information from the system.conf is available otherwise.

obbardc, 2020-03-05:
So using [your branch](https://github.com/ejoerns/rauc/commits/topic/slot-status-concept) we could use `busctl` inside the bundle or hook to get the system version ?

That would be great, what's the plans to merge that, could I help at all ?

obbardc, 2023-07-18:
@ejoerns Did you ever figure something out here? I've not really paid much attention to newer RAUC versions.

jluebbe, 2023-07-18:
We have the necessary information in the slot status file (ideally in a shared data partition with `data-directory=/data/rauc` or `statusfile=/data/central.raucs`). That information is already available via D-Bus.

With the refactoring in https://github.com/rauc/rauc/pull/1115 it should be easy to expose the necessary information as environment variables to hooks and handlers in a consistent way. For example:

```
RAUC_SLOT_BUNDLE_VERSION_<N>=2023.2
RAUC_SLOT_BUNDLE_BUILD_<N>=20230523123000
…
RAUC_BOOTED_SLOT_IDX=<N> # which would point to the slot information above
```

Would that be useful in your case?

obbardc, 2023-07-18:
Sounds perfect! I will retest to make sure that would work in my case.

jluebbe, 2023-07-18:
The best place to add it would likely be in `prepare_environment()` here: https://github.com/rauc/rauc/blob/master/src/install.c#L586

jluebbe, 2024-05-13:
@obbardc Perhaps #1400 can also handle your requirements?