rauc/rauc #534: D-Bus API BootSlot intended output

obbardc, 2020-02-06:
From testing it seems that `BootSlot` returns the device path, but from the documentation it sounds like it should return the name of the slot.

Is this intended behaviour?
Should I update the document?
Is it worth adding a BootSlotName property?

```
$ busctl get-property de.pengutronix.rauc / de.pengutronix.rauc.Installer BootSlot
s "/dev/mmcblk0p6"
```

ejoerns, 2020-03-04:
The 'BootSlot' property contains the information that is used to determine the booted slot.
It is derived from the kernel command line.
This can either be the slot name e.g. for `rauc.slot=rootfs.0` or the root device path e.g. for `root=PARTUUID=0815`.

This is more or less intended. Actually, it would probably be more helpful to actually get the slot name. And I'd agree that the documentation might be misleading here, yes.
Thus probably another property that actually returns the slot name could be an option.

What is your use case for obtaining the `BootSlot` property?

obbardc, 2020-03-04:
From testing I found the `BootSlot` property always returned the name of the booted slot (resolved from kernel cmdline by `get_cmdline_bootname()`), not the device path.

I will prepare a patch to update the documentation snippet to make this really clear.

I was thinking whether it would be worth adding a `BootSlotPath` property to point to the dbus service but my use-case doesn't need that at all yet.

for your interest, in my application we loop over the result of `GetSlotStatus`, finding the child containing `BootSlot` to get the version of the current booted system.

ejoerns, 2020-03-04:
> From testing I found the `BootSlot` property always returned the name of the booted slot (resolved from kernel cmdline by `get_cmdline_bootname()`), not the device path.

This is only valid if you have `rauc.slot=` or `bootchooser.active=` set from kernel commandline (which is recommended anyway). Then it returns the slots *bootname*, yes. But if you drop into  https://github.com/rauc/rauc/blob/master/src/context.c#L57 it will contain a `/dev/disk-by...` path.

> for your interest, in my application we loop over the result of `GetSlotStatus`, finding the child containing `BootSlot` to get the version of the current booted system.

This sounds pretty much like in-flight patch https://github.com/ejoerns/rauc/commit/7472c0686542c5c4847896db578b4bc1d2103f92

obbardc, 2020-03-04:
Ha, so it looks like I got it the wrong way around... I need some more coffee.

PR #550 (2020-03-06): docs/reference: clarify wording of BootSlot property contents
Closes: #534
closed 2020-03-06 by ejoerns