rauc/rauc #544: rauc info not working: Not enough substeps: check_bundle

lab-w, 2020-02-27:
Hello,

I try to use rauc to update a custom embedded Linux board. For that I use meta-rauc in a Yocto build flow.
When I try to get the bundle information on the target device by typing 
`rauc info filename.raucb`
 I get the following error message:
```
rauc-Message: Reading bundle: /home/root/filename.raucb

(rauc:2472): rauc-ERROR **: Not enough substeps: check_bundle (0/1)
Trace/breakpoint trap
```

Do you have any ideas what could be the cause?
Thanks!

ejoerns, 2020-02-27:
@lab-w seems as if you run into https://github.com/rauc/rauc/pull/519

Please check your `--keyring` argument.

Regards, Enrico

lab-w, 2020-03-02:
Thank you @ejoerns ! That was the right hint. I had not passed any `--keyring` argument - thinking the default path will be used. With `--keyring` and the absolute path it works.
A better error message could help others.

closed 2020-03-02 by lab-w

ejoerns, 2020-03-03:
@lab-w what do you mean with 'a better error message?'

The one above is not a question of error message but of incorrect code handling that leads you in a very different code path. With the patch from #519 you will get an output like:

```
140258319300624:error:02001002:system library:fopen:No such file or directory:bss_file.c:175:fopen('/path/to/test/my-keyring.pe','r')
```

This is note the most catchy hint as it is an error coming out of OpenSSL, but at least gives you the right information.

ejoerns, 2020-03-03:
See #547 for the outcome of a closer look at why this actually did not return a valid error code and message. Thus: Thanks for the indirect hint :smirk: :+1:

lab-w, 2020-03-03:
Happy to gave this indirect hint ;-)

I think I'm working with an old version of rauc (much before #547 ) and so the only output I got from the tool are the lines in my fist post. From that I had no idea that the issue has something to do with the path to the keyring.
So, somehow my fault - I'll update to a more up-to-date version.
But anyway: thanks for the fast support!