rauc/rauc #545: rauc convert should pass on chunk size parameter to casync

obbardc, 2020-02-28:
Currently casync bundles are built with the default casync chunk parameters, which currently is to build chunks with an average size of 64K.

This leaves us with lots of small chunks, when downloading over http with a lot rtt this can cause long delays.

Do you think this is worth passing this option through to casync?

ejoerns, 2020-02-28:
IMHO there is nothing that speaks against adding a configuration option for this.

obbardc, 2020-02-28:
ack, will try to put it on my backlog ;-)

jluebbe, 2020-02-28:
You can take a look at https://github.com/rauc/rauc/pull/537 for how this was done for mksquashfs.

ejoerns, 2020-02-28:
This raises the question if this is common enough to add a distinct configuration option, such as `chunk-size` or if this should also go into something like `casync-args`.

jluebbe, 2020-02-28:
> This raises the question if this is common enough to add a distinct configuration option, such as `chunk-size` or if this should also go into something like `casync-args`.

I'd prefer a `--casync-args` option, instead of passing them through separately. It won't be long until someone wants to try different compression options. ;)

ejoerns, 2020-02-28:
That `--casync-args` would anyway be required is clear. But we should provide distinct options for the common configuration parameters, like we have for `storepath` and `tmppath`. But, maybe the generic one is sufficient for this case first of all.

EugenWiens, 2020-02-28:
I think that the chunk size is a generic problem. This parameter would help
to setup better OTA updates regarding the bandwidth.

obbardc, 2020-02-28:
Personally I think it would be best to add `--casync-args` personally for potential future casync changes.

PR #546 (2020-03-10): Allow custom casync arguments with --casync-args
I left the test as a separate commit, not sure whether it gives full coverage? Let me know if changes needed or you would like me to squash.

Thanks!

Closes #545
closed 2020-03-10 by jluebbe