rauc/rauc #553: rauc fails to update when using busybox mount

jkroon81, 2020-03-07:
Using meta-rauc and yocto master branches, updating fails if "mount" is provided by busybox.

root@cm-itc:~# rauc install /tmp/bundle-cm-itc.raucb 
rauc-Message: 11:20:46.917: Active slot bootname: A
startedChecking and mounting bundle...rauc-Message: 11:20:46.927: Reading bundle: /tmp/bundle-cm-itc.raucb
rauc-Message: 11:20:46.930: Verifying bundle... 
rauc-Message: 11:20:47.167: Mounting bundle '/tmp/bundle-cm-itc.raucb' to '/mnt/rauc/bundle'
mount: mounting /dev/loop0 on /mnt/rauc/bundle failed: Invalid argument

(rauc:470): rauc-WARNING **: 11:20:47.173: Failed mounting bundle: failed to run mount: Child process exited with code 255
Failed mounting bundle: failed to run mount: Child process exited with code 255finishedInstalling `/tmp/bundle-cm-itc.raucb` failed

This setup works fine in thud.

If I install util-linux-mount package, then the update succeeds.

Should rauc require util-linux-mount, or can rauc be made to work with busybox "mount" ?

jkroon81, 2020-03-13:
I missed it, but the problem is printed to the console, it is the passing of the "sizelimit" parameter in src/mount.c that busybox mount doesn't like. If I remove that parameter, the update proceeds fine.

jluebbe, 2020-03-13:
Hmm, our docs say that busybox mount should be enough, but looks like that's not correct.
The intention of the sizelimit parameter is to make sure that only the explicitly checked part of the bundle is visible to squashfs. Maybe add support for it to the busybox mount applet?

ejoerns, 2020-03-13:
@jkroon81 I just had a look at the busybox code and apparently there does not seem to be any support for the `sizelimit` option. Neither per default nor configurable.

This would mean that busybox mount (in contrast to what the documentation suggests) is inappropriate for RAUC.

ejoerns, 2020-03-13:
Oups :)

jkroon81, 2020-03-15:
Relevant mail thread on LKML:
https://lkml.org/lkml/2019/11/30/83

jkroon81, 2020-03-15:
I can confirm that the suggested patch in
https://lkml.org/lkml/2019/12/10/1780
also successfully works around the problem.

jkroon81, 2020-06-25:
I don't see any solution to this emerging, so would you accept a patch to meta-rauc, adding util-linux-mount to rauc RDEPENDS ?

ejoerns, 2020-06-25:
@jkroon81 This would only solve it for the yocto integration.

We currently see this issue in a project too and also sketched a possible solution to fix this in busybox directly. If that should fail, adding the dependency to meta-rauc and mentioning in the documentation to not use busybox mount would be required, yes.

strumtrar, 2020-10-15:
This should be fixed with https://git.busybox.net/busybox/commit/id=6561e07460acc5b38af99aa5e80283dc04953eca

closed 2020-10-15 by ejoerns