rauc/rauc #554: Streaming bundle over http (withou casync)

mullerch, 2020-03-09:
Many devices don't have enough memory space to store the whole image of the system. In that case, the integrator must choose between either non-OTA updates (with external storage) or use casync.

As an alternative, it would be a real plus to have the possibility to update by streaming the bundle through http.

Is that compatible with the RAUC architecture and technologies used?

jluebbe, 2020-03-10:
> Many devices don't have enough memory space to store the whole bundle. In that case, the integrator must choose between either non-OTA updates (with external storage) or use casync.

Yes, that's the current situation. Is there a specific reason, for not using casync in your case?

> As an alternative, it would be a real plus to have the possibility to update by streaming the bundle through http.
> Is that compatible with the RAUC architecture and technologies used?

A normal RAUC bundle looks (mostly) like this: `[ squashfs ][ CMS signature over hash of squashfs ]`. To authenticate the squashfs before mounting, we need to calculate the hash over the full image, which is incompatible with streaming. In the casync bundle case, the squashfs contains the casync index, which is only relatively small.

For streaming, we would need to have on-demand authentication of individual blocks. A solution to this could look similar to what I suggested for [encryption](https://www.mail-archive.com/rauc@pengutronix.de/msg00086.html):
`[ squashfs hashed with dm-verity and optionally encrypted in dm-crypt compatible
format ][ CMS encrypted and signed container for (algorithm settings, dm-verity root
hash, payload key) ]` (although encryption would be optional). With that bundle layout, the client would download and authenticate the CMS container, and then fetch parts of the squashfs on-demand, authenticating them via the dm-verity hash tree.

To implement that, we'd need:
- [ ] write support for this new bundle layout (dm-verity, meta-data in CMS container)
- [ ] a proxy helper process for NBD (network block device) to HTTP(S) range-requests
- [ ] integration to authenticate and mount the new bundle layout  (mostly in `src/bundle.c`)

So far, it doesn't seem that this feature was important enough yes that someone implemented it themselves (or contracted us to). We'd be happy to review any PRs in this direction, of course.

ejoerns, 2021-01-08:
Note that the first part of this is exactly what was now implemented as the new bundle format in RAUC 1.5 [1]. Thus it brought us a significant step further towards native streaming support in RAUC.

[1] https://rauc.readthedocs.io/en/latest/reference.html?#verity-format

om26er, 2021-06-21:
Is there a way to track the current roadmap of RAUC somewhere ?

ejoerns, 2021-06-21:
@om26er No, there's no official (central) roadmap. Some general ideas we have in our brains, some are logical consequences about lessons learned or better understanding on how things should be.
But in the end, most of the development is actually driven by community or customer requirements. Some specific topics you can find in our issue tracker. We've marked them as "[RFE](https://github.com/rauc/rauc/issues?q=is%3Aopen+is%3Aissue+label%3ARFE)". If you are interested in these, feel free to discuss or to implement or to ask a consultant (like Pengutronix) to do so.

The verity format for example originated from our will to react on CVE-2020-25860 with something (additional) that is more than a mitigation of the drawback of the existing approach. Streaming support in turn is a logical consequence of the ability to verify bundle data 'on-the-fly'.

casync originally kicked in when we first had the demand to implement streaming and binary delta support in RAUC for a customer. As casync already implemented all we needed these days (and even more), we decided to use it and integrate it seamlessly  into the existing bundle format instead of reinventing the wheel ourselves.

jluebbe, 2023-08-10:
HTTP streaming is supported since [version v1.7](https://github.com/rauc/rauc/releases/tag/v1.7), see https://rauc.readthedocs.io/en/latest/advanced.html#http-streaming for details.

closed 2023-08-10 by jluebbe