rauc/rauc #556: RAUC verification with  Certificate Revocation List

djamesmec, 2020-03-10:
Hello,
I have created a bundle, signed with autobuilder-2.pem( revoked) from [test CA folder ](https://github.com/rauc/rauc/tree/master/test) .  I have installed dev-only-ca.pem as Keyring file on my target. 
 As per the script ([openssl-ca.sh](https://github.com/rauc/rauc/blob/master/test/openssl-ca.sh), the  dev-only-ca.pem must contain CRL with revoked certficate info. 
But it is observed that neither the "rauc info" cmd nor the "rauc install" cmd showed verification failure as "Certificate Revoked", with the bundle signed using revoked key.  "rauc info " showed the complete certificate chain. "rauc install" installed the bundle succesfully . 
rauc versions used :   rauc-1.0 and rauc-1.2.91
Could you let me know what went wrong here .

ejoerns, 2020-03-12:
Hi @djamesmec after a short investigation I can say that the X509 store in RAUC currently does not have the CRL checking flags set. Thus, at the moment, verification process does not take CRLs into account.
It's also not that easy as simply switching on the required flags as OpenSSL then enforces the presence of CRLs in a keyring/x509 store. This will surely break existing scenarios.

Thus our approach now is to add a system.config `[keyring]` option named `check-crl` which can be used to enable CRL checking.

As an additional safety net we also plan to add a check that should detect currently used (but still ineffective CRLs) by scanning the keyring files for a CRL header and emit a warning if a CRL is deceted but `check-crl` is not enabled.

Currently only some additional tests are missing to make a PR.

Most people anyway don't seem to use CRLs but do revocations by simply deploying a new keyring file with an update. Having CRLs will probably mostly have a benefit if they are retreived independent from the normal update process.

Thanks for reporting! 

Regards, Enrico

ejoerns, 2020-03-12:
I've pushed #561 now as a proposal to fix this.

djamesmec, 2020-03-18:
Hello @ejoerns, 
I have tested these changes and it worked . Thanks for the support.

jluebbe, 2020-03-18:
@djamesmec We'd be interested to know why using revocation instead of rotating the keyring is preferable in your use-case.

djamesmec, 2020-03-18:
@jluebbe We are targeting to pursue with a PKI hierarchy which includes Root CA, Intermediate CA and Signing CA levels. We are also intending to use different Signing CA for different set of devices. We are planning to keep our  Root CA Private Key offline as long as possible. So if a compromising of a particular signing key happened, we can revoke it, create a CRL and append it to the Keyring file without bringing root CA private Key online. I hope I answered your question .

jluebbe, 2020-03-18:
So you are using a separate key (from the CAs) to sign the CRL? Otherwise you would need to have access to the same (offline) keys to update the CRL anyway.

Does this work with OpenSSL as is?

djamesmec, 2020-03-18:
I will be using Intermediate CA  Private Keys to sign the CRLs if Signing key is revoked. For this case, as you mentioned, I have to access the Intermediate Private Key (offline) . 

OpenSSL verification(openssl command  with -crl_check  option ) of an image signed with Revoked signing Key, showed "certificate revoked" as expected.

jluebbe, 2020-03-19:
I'd like to have tests which cover this use-case. If you have time, could you provide an example (maybe as a github gist, maybe based on test/openssh-ca.sh) which sets up a PKI with the required keys and revokes a cert?

Are you using the indirect CRL feature or not?

PR #561 (2020-03-19): Allow to Enable CRL handling in RAUC's Bundle Verification
In order to actually use a provided CRL for checking, some flags need to
be set in OpenSSL. These were unset up to now. As a result of this, a
keyring / cert store file or directory could have contained a CRL, but
the Bundle verification process in RAUC simply ignored it.

This patch now introduces the `[keyring]` section option `check-crl` that
allows enabling CRL handling for RAUC's signature verification.

The reason why this cannot be enabled by default is that OpenSSL would
enforce having a CRL in a cert store. This will not only  [... cut]
closed 2020-03-19 by jluebbe