rauc/rauc #562: error after loop module is loaded

obbardc, 2020-03-16:
hi,
I have an error like this on the first system boot, mount fails with code 32 (and no further error). The second time the command is run it works OK.

I think it's due to loading loop module then attempting to setup the loop device so quickly.

```
Mar 16 07:24:23 localhost rauc[661]: Verifying bundle... 
Mar 16 07:24:23 localhost rauc[661]: Mounting bundle '/tmp/_download.raucb' to '/run/rauc/bundle'
Mar 16 07:24:23 localhost kernel: loop: module loaded
Mar 16 07:24:23 localhost rauc[661]: mount: /run/rauc/bundle: failed to setup loop device for /tmp/_download.raucb.
Mar 16 07:24:23 localhost rauc[661]: Failed mounting bundle: failed to run mount: Child process exited with code 32
Mar 16 07:24:23 localhost rauc[661]: installing https://fs.local/staging.raucb: Failed mounting bundle: failed to run mount: Child process exited with code 32
Mar 16 07:24:23 localhost rauc[661]: installing https://fs.local/staging.raucb: finished
Mar 16 07:24:23 localhost rauc[661]: installing `https://fs.local/staging.raucb` failed: 1
Mar 16 07:24:23 localhost system-bridge-d[743]: Failed mounting bundle: failed to run mount: Child process exited with code 32 (1)
Mar 16 07:24:23 localhost kernel: loop_set_status: loop0 () has still dirty pages (nrpages=1)
```

obbardc, 2020-05-15:
according to the hassos guys, this patch should fix it: https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/block/loop.c?id=7e81f99afd91c937f0e66dc135e26c1c4f78b003

ejoerns, 2020-06-16:
@obbardc That is something I haven't seen myself yet, but I have never used loop support as a module yet.
In my opinion, on an embedded system targeted to use RAUC, the SquashFS support should normally be built-in.

The patch you mention is now part of released Linux v5.7, thus we could add a note in the integration documentation that loop should not be used as a module before 5.7.

@obbardc Is there any special reason to use SquashFS as a module instead of having it built-in that we should have in mind?

obbardc, 2020-06-16:
That all sounds good to me. Some projects (like the linked HASSOS above)
use rauc with Debian and with the default kernel config, squashfs is loaded
as a module. Other than that, I can't see any problem with built-in
squashfs.
I'll add the documentation points mentioned to a MR at some point shortly.

ejoerns, 2020-06-18:
@obbardc Ok, for the Debian kernel I understand the issue. A MR for documentation sounds great.

Thanks!

PR #625 (2020-10-27): docs: include Kernel loop device patch
There is a bug in Kernel version v5.0 until v5.7 when loop device
support is loaded as a module, in a small percentage of cases
mounting the bundle fails. Include information about the patch.

Resolves: #562
Resolves: #582
commit 666fafd0 (2020-10-27): docs: include Kernel loop device patch

There is a bug in Kernel version v5.0 until v5.7 when loop device
support is loaded as a module, in a small percentage of cases
mounting the bundle fails. Include information about the patch.

Resolves: #562
Resolves: #582
closed 2020-10-27 by jluebbe