rauc/rauc #563: One bundle with rootfs and files

mmar2228, 2020-03-24:
Dear all,

I would like to have a single bundle with 3 images:

* rootfs.ext4
* dtb file
* kernel image file

My system has the following partitions:

* /dev/mmcblk0p1: (vfat) contains Image and my-dtb.dtb files for target group A 
* /dev/mmcblk0p2: (vfat) contains Image and my-dtb.dtb files for target group B 
* /dev/mmcblk0p3: (ext4) rootfs for target group A 
* /dev/mmcblk0p3: (ext4) rootfs for target group B 

Both "Image", "my-dtb.dtb" are intended to be updated with my bundle, just a copy would be enough. rootfs is also intended to be updated with the same bundle.

I am using Yocto with the following core-bundle-minimal.bb configuration:

```
RAUC_BUNDLE_COMPATIBLE = "my_machine"

RAUC_BUNDLE_SLOTS = "rootfs kernel dtb"

RAUC_SLOT_rootfs = "core-image-minimal"

RAUC_SLOT_dtb = "u-boot"
RAUC_SLOT_dtb[type] = "file"
RAUC_SLOT_dtb[file] = "my-dtb.dtb"

RAUC_SLOT_kernel = "linux"
RAUC_SLOT_kernel[type] = "file"
RAUC_SLOT_kernel[file] = "Image"

RAUC_KEY_FILE = "path-to-key/key.pem"
RAUC_CERT_FILE = "path-to-cert/cert.pem"
```

This recipe gives me a bundle without errors. However I have problems to install it. 

My system.conf is the following, as I need to copy "Image" and "my-dtb.dtb" into /dev/mmcblk0p1(2) and "rootfs.ext4" be written into /dev/mmcblk0p3(4).

```
[system]
compatible=my_machine
bootloader=uboot

[keyring]
path=/etc/rauc/cert.pem

[slot.rootfs.0]
device=/dev/mmcblk0p3
type=ext4
bootname=A
resize=true

[slot.rootfs.1]
device=/dev/mmcblk0p4
type=ext4
bootname=B
resize=true

[slot.kernel.0]
parent=rootfs.0
device=/dev/mmcblk0p1
type=vfat

[slot.kernel.1]
parent=rootfs.1
device=/dev/mmcblk0p2
type=vfat

[slot.dtb.0]
parent=rootfs.0
device=/dev/mmcblk0p1
type=vfat

[slot.dtb.1]
parent=rootfs.1
device=/dev/mmcblk0p2
type=vfat

```

However, I get the following error:

>  LastError: Installation error: No target slot for class dtb of image my-dtb.dtb.img found

Any idea?

I think I have any misconfiguration related to kernel Image and my-dtb.dtb files, because they both must be copied into the same partition /dev/mmcblk0p1 (for updating bootname A).

Please, I will kindly appreciate your help.

Regards,

MM

ejoerns, 2020-03-24:
Hi @mmar2228 

this won't work as you have to have a single image for the entire slot. Not fully sure why it complains about 'dtb' slot here but it will surely fail anyway.

The way to handle this would be to have only the 'kernel' slot and generate a vfat image containing both the kernel and the dtb. This could be done with a script, using wic or genimage.

Note that in many cases it is sufficient to have the kernel and dtb in the rootfs. But this depends a little on your platform and your bootloader.

The normal way to create such a 'split out' image would be to build it into e.g. /boot folder of rootfs first of all and then let wic or genimage split the content of this folder out into a separate image.

Hope that helps, best regards, Enrico

mmar2228, 2020-03-24:
Thanks you for your useful help.

I have changed the core-bundle-minimal.bb configuration:

```
RAUC_BUNDLE_COMPATIBLE = "my_machine"

RAUC_BUNDLE_SLOTS = "rootfs kerneldtb"

RAUC_SLOT_rootfs = "core-image-minimal"
RAUC_SLOT_rootfs[file] = "core-image-minimal.ext4"

RAUC_SLOT_kerneldtb = "core-image-minimal"
RAUC_SLOT_kerneldtb[type] = "image"
RAUC_SLOT_kerneldtb[file] = "kernel.img"

RAUC_KEY_FILE = "path-to-key/key.pem"
RAUC_CERT_FILE = "path-to-cert/cert.pem"
```

In addition I have created a core-image-minimal.bbapend in order to create the desired image containing the Image and dtb files:

```
do_image_complete_append(){
    os.system('dd if=/dev/zero of=kernel.img bs=1M count=50')
    os.system('printf "o\nn\np\n1\n\n\nt\nc\nw\n" | fdisk kernel.img')
    os.system('mkfs.vfat kernel.img')
    os.system('mcopy -i my_dtb.dtb :$
    os.system('mcopy -i kernel.img Image ::Image')
}
```
However, bitbake core-bundle-minimal gives me a wrong raucb file, as I get the following "rauc info" (second image is not the desired kernel.img indicated in bundle recipe):

```
2 Images:
(1)	core-image-minimal.ext4
	Slotclass: rootfs
	Checksum:  e2edb11b40095986977dafc58fd9c088c855eb49b5cd0530ba899736fc4834f5
	Size:      687865856
	Hooks:     
(2)	core-image-minimal.ext4
	Slotclass: kerneldtb
	Checksum:  e2edb11b40095986977dafc58fd9c088c855eb49b5cd0530ba899736fc4834f5
	Size:      687865856
	Hooks:     
0 Files
```

Why do I obtain this wrong rauc bundle file?

Thanks in advance,

MM

ejoerns, 2020-03-26:
@mmar2228 more moving to be a meta-rauc question now, but as far as I can see, bitbake does not care for the slot variable varflags when handling vardeps.

I assume you've built the recipe before when the `RAUC_SLOT_kerneldtb[file]` was not set, yet?

Please try to force a rebuild, e.g. with `bitbake -c cleanall -f <bundle-recipe> && bitbake <bundle-recipe>` and see if the problem occurs.

For now, my bitbake skills are at the end and it would require deeper investigation if that is expected behavior.

Enrico

mmar2228, 2020-03-31:
Hello,

Yes, I cleaned recipe before rebuilding it. But result still is the same.

Finally, as I am not able to get my bundle by means of recipes in Yocto build process, I create the desired bundle after build process with a bash script, using `rauc bundle` command and an hand-written manifest. It is an partial solution but it works.

Thanks very much for your interest and replies.

MM

Avni22, 2022-01-21:
Hi, 
@mmar2228 

I am working on same and I try to modifi u-boot for RAUC A/B partition, now kernel & dtb is not able to load after boot
and it give error for not able to load DTB & image.  

can you please share what are the changes need in u-boot, if possible? bash script for generating rauc bundle. 

I will kindly appriciate your help. 

Thanks

ejoerns, 2022-01-21:
> Hi, @mmar2228
> 
> I am working on same and I try to modifi u-boot for RAUC A/B partition, now kernel & dtb is not able to load after boot and it give error for not able to load DTB & image.

Did you have a look at the integration documentation?

https://rauc.readthedocs.io/en/latest/integration.html#id5

Also note that u-boot configuration highly depends on your actual storage, partitioning, etc. Thus unless we do not have more details it is impossible to give general advice.

> can you please share what are the changes need in u-boot, if possible? bash script for generating rauc bundle.

The purpose of the BSP  is to build properly usable artifacts, thus if there is an issue in the build system we should fix this instead of creating hacky scripts around it.

I assume you are using Yocto with meta-rauc?

> I will kindly appriciate your help.
> 
> Thanks

Avni22, 2022-01-24:
Hi,

Thank you for your replying. 

I am using meta-rauc for target board only not for creating bundles. 
for creating bundle i am following this [https://github.com/rauc/rauc].

I am facing issue during updating rauc bundle , can you me help to look what are the issue? 
Attaching the logs during updates, and menifest.rauc, system.config data below.

RAUC Logs :  
--------------------------------
rauc[344]: Verifying bundle signature...                   +-----------------------------+
rauc[344]: Verified detached signature by 'O = Test Org, CN = Test Org Development-1'
rauc[344]: Mounting bundle '/home/root/otgPD/rootfs_test.raucb' to '/run/rauc/bundle'
rauc[344]: Configured loop device '/dev/loop0' for 4096 bytes
rauc[344]: Starting pre install handler: /usr/lib/rauc/preinstall.sh
rauc[344]: incompatible manifest for this system (IMX_IMX8QXP): IMX_IMX8QXP 
rauc[344]: Installation error: Compatible mismatch
rauc[344]: installing /home/root/otgPD/rootfs_test.raucb: Installation error: Compatible mismatch
rauc[344]: installing /home/root/otgPD/rootfs_test.raucb: finished
rauc[344]: installing `/home/root/otgPD/rootfs_test.raucb` failed: 

manifest.raucm
-------------------------------
[update]
compatible=IMX_IMX8QXP
version=1.000

[image.rootfs]
sha256=af963c259132189874b49a499d55d12c7ae75d3b094464e38aa98e85722e1fe4
size=1944
filename=rauc.tar.xz

system.config
-----------------------------
[system]
compatible=IMX_IMX8QXP
bootloader=uboot
statusfile=/home/root/central_slot.raucs

[keyring]
path=ca.cert.pem

[handlers]
pre-install=/usr/lib/rauc/preinstall.sh
post-install=/usr/lib/rauc/postinstall.sh

[slot.rootfs.0]
device=/dev/mmcblk0p2
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/mmcblk0p3
type=ext4
bootname=B

[slot.appfs.0]
device=/dev/mmcblk0p4
type=ext4

ejoerns, 2022-01-26:
@Avni22 As this is not an issue but a question (as far as I can tell) and unrelated to the original issue, would you please open a separate discussion here: https://github.com/rauc/rauc/discussions ?

Your installation fails because of a compatible mismatch, btw. Thus please check that this is actually consistent and no leading/trailing whitespace is hidden somewhere.

closed 2023-10-24 by ejoerns