rauc/rauc #565: systemd-boot support avoiding EFI variables

mhill-holoplot, 2020-03-27:
We are developing a product with a sadly proprietary third party EFI implementation. We are aware that some bad EFI implementations have a tendency to brick systems if EFI variables are updated too much. Naturally we will try to evaluate the EFI implantation on our hardware, but we are considering using sd-boot's file based approach avoiding the problem all together.

I'm referring to the system described here: https://systemd.io/AUTOMATIC_BOOT_ASSESSMENT/

I've started down the route of putting an implementation together, so I'm wondering whether anyone else has looked into doing this and/or has wisdom to share?

ejoerns, 2020-03-27:
@mhill-holoplot I'd really appreciate having support for `systemd-boot`. We also have this on our wish list but so far it was not backed by any actual project that required this, yet.
It should be a good alternative to this bad EFI stuff.

This should be something that could be integrated into the rauc boot selection interface but I haven't looked into the requirements and capabilities of systemd-boot deeply enough to give any useful hint on this side of the implementation.

Maybe anyone else already did?

mhill-holoplot, 2020-03-27:
Unfortutalty there is no C API exposed from systemd for this there is `bootspec.h`, but it's read only and not some think I think can be easy used outside of the systemd repo.

The main path that seems to be open is parsing everything from the file system within RAUC. Gets a bit more complicated because the default field in `loader.conf` can be a pattern. I still need to poke around a bit to figure out sd-boot's exact behaviour in regards to implementing set/get primary functions.

Other question is whether an API for sd-boot could be added to systemd. There appears to be a dormant issue concerning this https://github.com/systemd/systemd/issues/11221

jluebbe, 2020-04-21:
Some other systemd tools (systemctl, busctl) have support for json output, but bootctl doesn't support that yet. The bootloader backends for barebox, u-boot and grub already run the respective userspace tools, so I'd take a PR adding support by calling and parsing bootctl.

ejoerns, 2020-06-16:
@mhill-holoplot didi you make progress with this topic? Anything we could support you with?

mhill-holoplot, 2020-06-17:
I got started on the systemd work. But didn't get very far with it before having to tackle some more pressing issues. I'd like to get back to it but right now I can't estimate when.

The JSON approach @jluebbe suggested seems a bit more likely to get a quick merge in systemd than extending the API. I'd be quite happy to go down that route if parsing JSON from stdout is acceptable here.

poettering, 2020-08-20:
@zonque asked me to comment here, from a systemd-boot upstream perspective. 

What precisely are you looking for in an API from systemd/systemd-boot upstream?

1. is it about generating boot loader spec entries? if so, we don't even have that upstream, there's just the kernel-install script we ship upstream, which does that from shell
2. or is this about parsing boot loader spec entries? we do have that, but I am not too happy to export that, it's not written for that, and would need some serious love first.
3. or is this about installing/updating sd-boot in the ESP?

So if you are looking for 1, we have nothing really. but given how simple the boot loader spec is I think it should be trivial to reimplement this in your own code. the spec is written to be trivial, easily reimplementable, and I think that'd be the way to go. See https://systemd.io/BOOT_LOADER_SPECIFICATION for details.

If you are looking for 3: a long time ago we had the plan to expose what "bootctl install" and "bootctl update" do as bus API, but I am not convinced this is desirable, since D-Bus is mostly about opening up priv ops to unpriv clients, but i figure an updater for an OS is likely as privileged as bootctl would have to be. My current way of thinking is simply making the bootctl cmdline the API to hack against: fork it out and wait for it to finish. In order to have parsable info back about errors and such you can set up `$NOTIFY_SOCKET` IPC, i.e. the sd_notify() API that system services can use to do ready notification to systemd, but it's actually a carrier for all kinds of other messages. You already get slightly better error reporting back if you set that up when invoking bootctl. I think we can improve that further, by adding progress info and such.

I also intend to add a service to systemd that invokes "bootctl update" on every boot, so that you can off-line update /usr, and it will propagate into the ESP as needed automatically the latest when it becomes activated. Since "bootctl update" is idempotent and a NOP if the ESP is already up-to-date this should be safe.

But anyway, we can certainly add more hook-up points in systemd upstream, but I'd really like to know what exactly is needed here, as I do not know rauc. Hence, please let me know the precise, minimal operations you#d like us to support as an API

poettering, 2020-08-20:
(btw, for embedded devices I'd actually suggest using boot loader spec type `#2` entries instead of type `#1` snippets. i.e. build unified kernels, that include the initrd and meta info, and can be installed/updated in one file. It makes things more robust, and is nice for secure boot since the firmware can verify the whole thing in one go.)

jameshilliard, 2020-08-27:
I'm not using rauc, but I have the same issue with [swupdate](https://sbabic.github.io/swupdate/) which should be directly applicable here.

Currently I have a hacky lua script that uses the bootctl binary for installation and setting/reading default loader entries, however this is far from ideal as the command line output from bootctl is not stable.

Here's what I think would work the best for now:

Split all bootctl operation functionality out into a library, say called `libboot` or something, so that any updater apps can easily do everything bootctl is capable of simply by using the `libboot` library. The bootctl cli binary would then use the functions from this library for all operations.

This would allow doing equivalent operations to `bootctl list`, `bootctl install`, `bootctl set-default` or `bootctl update` just by having the updater use the `libboot` library functions. This is basically how we handle the uboot integration in swupdate, we just include [libuboot.h](https://github.com/sbabic/libubootenv/blob/master/src/libuboot.h) and use the functions exposed there.

> 1. is it about generating boot loader spec entries? if so, we don't even have that upstream, there's just the kernel-install script we ship upstream, which does that from shell

While not as big an issue it would be handy if this functionality was available in a `libboot` style library.

> 2. or is this about parsing boot loader spec entries? we do have that, but I am not too happy to export that, it's not written for that, and would need some serious love first.

Yeah, I do need this functionality as well, currently I parse it from the `bootctl list` output within my lua script.

> 3. or is this about installing/updating sd-boot in the ESP?

This is also functionality I need, which I currently handle by calling `bootcli install` within my lua script.

> (btw, for embedded devices I'd actually suggest using boot loader spec type `#2` entries instead of type `#1` snippets. i.e. build unified kernels, that include the initrd and meta info, and can be [...]

I don't think this is generally workable on many types of embedded systems.

From the description:
> A unified kernel image is a single EFI PE executable combining an EFI stub loader, a kernel image, an initramfs image, and the kernel command line.

It appears that an initramfs image is required, which is something that outright doesn't exist at all in many embedded Linux setups. Usually what we have in the embedded world is either a ramdisk only rootfs or no ramdisk whatsoever, my current setup does not use a ramdisk at all.

Another issue with using boot loader spec type `#2` for a setup like mine is that I need to generate the kernel command line at installation time, not at build time in order to correctly specify the rootfs `PARTUUID`