rauc/rauc #566: Is RAUC compatible to TUF specification?

lab-w, 2020-03-30:
I'd like to know if there are any intentions to make RAUC compatible to TUF specification (https://theupdateframework.io/).
Are there any use cases where TUF and RAUC are used in combination? In my understanding this would double some steps because RAUC also cares about signing of the bundles.
Or maybe there are good reasons to do not combine/integrate TUF and RAUC?
Maybe some of the developers/users of RAUC already have experience with TUF. I'm interested in your thoughts.

jluebbe, 2020-04-20:
No, RAUC is not compatible to the TUF spec and I don't know of cases where they are used in combination.

During the initial design, we looked at TUF and decided that the complexity would be too high for most use-cases, which only have a one or very few people who build, sign and deploy the updates. Using a HSM to store those keys also gives high level of protection against key compromise without the complexity of TUF.

TUF also has some features which don't really apply to offline updates (via USB memory stick) or image-based full system updates, but they still would increase the complexity of the implementation.

Some aspects like freshness (timestamps) or requiring multiple (N of M or specific) different signatures could be implemented in RAUC without significant changes, as they are already covered by the CMS signature format we use. It would mainly need some handling of the details via the OpenSSL API.

If you combine TUF with RAUC, maybe one of TUF's roles could be mapped to the bundle signature. Otherwise, you could have a "dummy" signature on the bundle. If there are performance bottlenecks caused by the redundant verification, I'd prefer to optimize RAUC's own checks, so also the non-TUF-users would benefit from that work.

Which specific features or properties of TUF are you interested in?

ejoerns, 2020-06-16:
@lab-w Did that answer your questions? Do you have any specific features you are interested in and would like to see supported?

lab-w, 2020-06-22:
Thanks @jluebbe and @ejoerns for the detailed answer. Currently I do not have a specific feature of TUF I'd like to see in RAUC - I was just interested in some opinions about it.

closed 2020-06-22 by lab-w