rauc/rauc #571: Updating keyring and slot restrected keys

loblik, 2020-04-10:
I have a following use-case. System is deployed with root CA and certificate for bundle verification. Then a other party involved is provided with certificate (issued by the root). This certificate is then installed to the system and can be used to upgrade some slots but not all of them.

This requires some mapping between certificate and allowed slots to update. Key usage attributes with custom OIDs could serve this probably.

From point of rauc it would need two things I guess:

- Be able to do verified keyring updates.
- Somehow involve slot names in verification process.

My question is:
- Would something like this be currently possible with rauc, using some hooks or something?
- Or would it be possible with some reasonable adjustments to the core (extending hooks capability for instance) ?
- Since signature is computed over the whole bundle, to be able to do an update where artifacts are signed by different keys would require big changes I guess.

jluebbe, 2020-04-20:
> * Would something like this be currently possible with rauc, using some hooks or something?

Currently not. Also keep in mind that RAUC is designed with the assumption that the bundle content is fully trusted after the certificate is checked. For example, hooks will be executed from the bundle with full privileges. Also we mount the bundle via the kernel squashfs implementation, which is not explicitly designed to handle untrusted input.

> * Or would it be possible with some reasonable adjustments to the core (extending hooks capability for instance) ?

Hooks run after signature verification and in full control by then.

> * Since signature is computed over the whole bundle, to be able to do an update where artifacts are signed by different keys would require big changes I guess.

Yes. How hard it would be depends on if this should actually be secure or just some sanity checks.

It's probably easier to have some automated service to check that a bundle signed by the other party contains only the allowed parts (and no hooks), and then have a way to resign that with a key trusted by the devices.

Alternatively, I've heard there as some deployments which use RAUC for the base system and ostree for additional applications.

ejoerns, 2020-06-16:
@loblik Did this answer your question? Did you find a method to limit installation capabilities for specific signatures?

loblik, 2020-06-26:
It seems that we will use something else (probably swupdate). Though I like the simplicity of rauc. However it seems it does not fit our case well. So we can close this.

ejoerns, 2020-06-27:
@loblik ok. thanks for the feedback!

It would always be interesting to hear about use cases we did not have in mind yet (if you would like to share your experience).

I'll close this first of all, but feel free to comment on it anyway if you like ;)

closed 2020-06-27 by ejoerns