rauc/rauc #581: Certificate is not yet valid

ArashJavan, 2020-04-24:
I jus tfollowd the example in doc and created an test key pair as written in doc:

```
> rauc-Message: 00:50:26.265: Reading bundle: /root/update.2020.04-1.raucb
> rauc-Message: 00:50:26.271: Verifying bundle...
> signature verification failed: Verify error:certificate is not yet valid
```
Am i doing something wrong?

ejoerns, 2020-04-24:
@ArashJavan Did you do it with

    openssl req -x509 -newkey rsa:4096 -nodes -keyout demo.key.pem -out demo.cert.pem -subj "/O=rauc Inc./CN=rauc-demo"

?

This can happen if the time on the machine you used for creating the certificate is more recent then the time on the machine where you installed it. Or is it the same machine?

ArashJavan, 2020-04-27:
@ejoerns Thanks for your reply, your were right, machins had diff. times.

closed 2020-04-27 by ArashJavan